Xilinx/BNN-PYNQ #6: Warning: Cannot find design file

jpan127, 2017-05-21:
Hi I am trying to setup this package and I am running into a few issues.

I cloned the repository, set the XILINX_BNN_ROOT variable, launched the hw shell script for both cnv-pynq and lfc-pynq but I get this error.

[image]

Another issue I have is I'm not clear on the quick start instructions for the Pynq board.  It is asking to copy the include folder from Windows  **to the** Pynq board correct?  I was wondering if there is an easy way to copy the folder since there is over 200 files, and Jupyter does not let me import folders, only files.

Thanks,
Jonathan

giuliogamba, 2017-05-23:
Hi Jonathan, 

you can access and move data in PYNQ using Samba as described here (section "Accessing files on the board"): 
https://pynq.readthedocs.io/en/latest/1_getting_started.html#using-pynq

About your first issue, the flow has been tested on Unix systems. Can you provide more information about the flow you are trying to use? Anyway, at a first look it seems that you should change the XILINX_BNN_ROOT path using slash "/" instead of backslash "\" .

jpan127, 2017-05-24:
Hi giuliogamba,

I am using Windows 10 64 bit, Vivado 2016.4 design suite, GCC 6.2.0.  I changed my environment variable to forward slashes instead of back slashes and that got rid of the first error.  However I am getting a second error with compilation.

[image]
[image]

closed 2017-07-20 by giuliogamba