Xilinx/BNN-PYNQ #9: After bitsteam generation

mgepner, 2017-07-20:
Once the bitstream is generated, do you program the FPGA with the bitstream using Vivado? How do you have the FPGA take an MNIST file and measure its performance? I have the bitstream created, as well as Linux running on the PYNQ and the mnist_parameters.npz/binparam-lfc-pynq created. What are the next steps?

giuliogamba, 2017-07-21:
In order to use your generated bitstream, you can follow step 6 in HW design rebuild available at 
https://github.com/Xilinx/BNN-PYNQ/blob/master/README.md.

The binary parameters should be moved in pip_installation_path/bnn/params/

mgepner, 2017-07-21:
Okay, I programmed the PYNQ now using the bit file found in bnn/bitstreams. The PYNQ is also running Linux. Are you saying the binary parameters in /bnn/params/ need to be moved onto the PYNQ?

giuliogamba, 2017-07-21:
Yes, the parameters are stored on the PYNQ board after pip installation, and if a new dataset is trained those should be moved in the same path (_pip_installation_path_/bnn/params/_)

mgepner, 2017-07-31:
The PYNQ is programmed with a bitstream through Vivado and has the necessary thres/weights files in /bnn/params/mnist now. How do you run the BNN on the PYNQ?

giuliogamba, 2017-07-31:
No, the PYNQ in not programmed through Vivado. Please refer to the "Using Overlays" paragraph in http://pynq.readthedocs.io/en/latest/16_creating_overlays.html.
To see how the BNN are run, please check the bnn.py class in the bnn subfolder

closed 2017-09-01 by giuliogamba