Xilinx/BNN-PYNQ #12: Undefined Latency and Interval values in Vivado HLS

MohammadSamragh, 2017-09-17:
When I run the "make-hw.sh" file, the compilation finishes successfully. However, Vivado HLS reports the Latency and Interval values as undefined (marked as "?"). I would appreciate if you let me know how I can get the throughput reports.

giuliogamba, 2017-09-18:
Hi, 

in order to provide an estimate, Vivado HLS needs to know how many input images will be analyzed. 
The number of images is currently passed as a runtime parameter (_numReps_) making the tool unable to predict latency. 
By fixing at compile time _numReps_, Vivado HLS will be able to estimate the latency and interval cycles.

MohammadSamragh, 2017-09-18:
Thank you very much for your response. I also had to slightly change the two functions "Mem2Stream_Batch" and "Stream2Mem_Batch" in the "dma.h" file as follows:
```
void Mem2Stream_Batch(ap_uint<DataWidth> * in,
		stream<ap_uint<DataWidth> > & out, const unsigned int numReps) {
	const unsigned int indsPerRep = numBytes / (DataWidth / 8);
	unsigned int rep = 0;
	// make sure Mem2Stream does not get inlined here
	// we lose burst inference otherwise

	const int iters_divisable_by_16=(int)(numReps/16);
	const int iters_remainder=numReps-iters_divisable_by_16*16;
	for (int i=0;i<iters_divisable_by_16;i++)
	{
		Mem2Stream<DataWidth, numBytes * 16>(&in[rep * indsPerRep], out);
		rep += 16;
	}
	for (int i=0;i<iters_remainder;i++)
	{
		Mem2Stream<DataWidth, numBytes>(&in[rep * indsPerRep], out);
					rep += 1;
	}
	/*while (rep != numReps) {
		unsigned int repsLeft = numReps - rep;
		if ((repsLeft & 0xF) == 0) {
			// repsLeft divisable by 16, read 16 images
			Mem2Stream<DataWidth, numBytes * 16>(&in[rep * indsPerRep], out);
			rep += 16;
		} else {
			// fallback, read single image
			Mem2Stream<DataWidth, numBytes>(&in[rep * indsPerRep], out);
			rep += 1;
		}
	}*/
}
```

closed 2017-09-18 by MohammadSamragh