Xilinx/BNN-PYNQ #17: Runing other BNNs on PYNQ using FINN

zhangqingbo1, 2017-11-09:
Hi! 
I have finished the PYNQ installing and ran the  python notebooks examples successfully. on the other hand ,in order to find how the BNN works on the PYNQ, I read the 'FINN' papar carfully,but still have many problems in the papar . I have two questions：
1. Are there futher way to study your FINN?
2. Can I Runing the  BNNs designed by myself  which have diffirent  topologies on PYNQ using FINN，and how to do it ?

giuliogamba, 2017-11-27:
Hi, 

if I've understood your question, you would like to build a new neural network topology on PYNQ. 
Currently the tool to automatically generate the HLS description of the neural network is not open-sourced, but you can write your own topology by using the HLS library provided in this  [repository](https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/src/library/hls) 
You can use the examples available for the [CNV](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/cnv-pynq/hw/top.cpp)  and [LFC](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/lfc-pynq/hw/top.cpp) topologies as a starting point to build your own neural network accelerator.

zhangqingbo1, 2017-11-28:
Oh Thanks ,yeah, I want to build a new neural network topology on PYNQ. I have read the the FINN hardware library code in bnn/src/library/hls,  it's a little difficult for me but I will try again. 
when write my own topology by using the HLS library,do I need to do the step "Theano + BinaryNet  ----->BNN topology&paramenters"?  
And  the topologies of CNV and LFC are determined  by the "Theano+BinaryNet"  or the HLS library provided in this repository?

giuliogamba, 2017-11-28:
Hi, 

yes, with Theano you will train the neural network with the topology you selected, and the same topology has to be implemented in the hardware accellerator.

zhangqingbo1, 2017-11-29:
Thankyou, I will try .