Xilinx/BNN-PYNQ #18: Environment setting

zhangqingbo1, 2017-11-24:
Hi,
 When doing the Hardware design rebuilt,  how can I set the XILINX_BNN_ROOT environment variable to clone_path/BNN_PYNQ/bnn/src/.  I am not good at Ubuntu ,please give an advice.

giuliogamba, 2017-11-27:
Hi, 

for sh and bash, you can use `export XILINX_BNN_ROOT=clone_path/BNN_PYNQ/bnn/src/`
for csh, you can use `setenv XILINX_BNN_ROOT clone_path/BNN_PYNQ/bnn/src/`

Please change clone_path to the root path in which you cloned the repository.

zhangqingbo1, 2017-11-28:
There are a new problem,after export XILINX_BNN_ROOT=clone_path/BNN_PYNQ/bnn/src/ ,terminal says vivado not found in path, I  set ubuntu's environment variable to vivado and vivado_hls, but still have the problem.  PLS give me some advice.

giuliogamba, 2017-11-28:
Hi, for this please refer at the [Vivado design hub](https://www.xilinx.com/support/documentation-navigation/design-hubs/dh0013-vivado-installation-and-licensing-hub.html), particularly to UG973.

zhangqingbo1, 2017-11-29:
OK  thanks!

closed 2017-11-29 by zhangqingbo1