Xilinx/BNN-PYNQ #19: Question about driver and host code

LiYang66, 2017-12-21:
Hi!

     I am a beginner to use PYNQ and I find BNN-PYNQ is a excellent demo for me. My goal is to write a new neural network accelerator which also using PYNQ platform, Now I understand the hls codes, however, the driver codes and host codes make me frustrated, So I want to ask that:
    1 how to write the driver codes and host codes In the LIBRARY folder(I have no background about ARM)
    2 what software did you use for driver codes and host codes. Because I know all the HLS codes  are used in the Vivado_HLS software, I don't know which software to implement driver codes and host codes and how to do debug and simulation of these codes. Did you use SDK or SdSoc? 
     3 I didn't see test bench codes in Vivado_HLS of the BNN-PYNQ. Does there has any way to simulate your work?

Thank you so much!  Looking forward to your reply!

giuliogamba, 2018-01-08:
Hi, 

In the last commit I've added the code for C simulation in Vivado HLS. Once you launch the make-hw script, the Vivado HLS project that will be produced will have the testbench available. 
For question 1) and 2), the [driver](https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/src/library/driver) folder list the functions for Axi communication with the PL and memory allocation. The addresses generated from vivado are then used to address the axi-lite registers, as depicted [here](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/library/host/foldedmv-offload.cpp).
The code is in C++ and to run it it's enough to compile with gcc. 

BR

GG

LiYang66, 2018-01-08:
Thank you so much!! You really help me a lot! @giuliogamba

closed 2018-01-08 by LiYang66