Xilinx/BNN-PYNQ #21: Questions about hardware design rebuild

bramblewalls, 2018-01-26:
Hi!
I am trying to rebuild the hardware design, but i find there are some problems.
My system is centos 7, and your recommend system is Ubuntu, i have no idea whether there is problem with system.My vivado version is Vivado 2017.4 webpack.
When i run the "make-hw.sh" file it says that can not find tiny_cnn. So i download tiny_cnn from https://github.com/Xilinx/xilinx-tiny-cnn. Then i copy the tiny_cnn folder to XILINX_BNN_ROOT/library/host.
I am not sure whether my tiny_cnn version is right and whether my tiny_cnn work well but problem disappear when i do like this however then new problem appear.
[image]
I am not very fimilar with Vivado HLS, please give me some advices.
Thanks a lot!

giuliogamba, 2018-01-26:
Hi, 

in the latest release, we are running c simulation by performing inference on a test image and checking the accuracy of the result. If csim fails, the HLS synthesis is not performed. 
I've just tried with 2017.4 and everything works on my side, so I have problems in reproducing the error. 
There are 2 warnings, can you please reply with the content of the warning?

bramblewalls, 2018-01-26:
Thanks for you reply!
There is the content of the 2 warnings
[image]

bramblewalls, 2018-01-26:
Thanks for your reply!
There is the content of the 2 warnings.
[image]

giuliogamba, 2018-01-26:
Please try to make a fresh clone with the new fix.

bramblewalls, 2018-01-26:
My problem is solved, thank you very much!

closed 2018-01-26 by bramblewalls