Xilinx/BNN-PYNQ #22: Running BNN-PYNQ on a ZYNQ board?

hex0102, 2018-02-11:
Is it possible to port this project to a zynq board?

giuliogamba, 2018-02-12:
Hi, 
the PYNQ-Z1 board, targeted in this repo, features a ZYNQ-XC7Z020. 
So the project is already targeting ZYNQ.

hex0102, 2018-02-13:
@giuliogamba Thanks very much for the response.  I'm using Zynq-7000 APSoC ZC702 Evaluation board. So I guess it should be OK. I also wonder if there're additional features on the PYNQ-Z1 board.

maltanar, 2018-02-13:
@hex0102 [my comments on a similar question](https://github.com/Xilinx/BNN-PYNQ/issues/10) might be helpful. In summary: this is indeed possible but you'll need to run the PYNQ Linux distribution on your ZC702 board to use the repo without any changes.

closed 2018-07-10 by giuliogamba

praveshgoyal, 2019-03-30:
Hi
@giuliogamba ,I am using  Zynq-7000 SoC, So will repo work fine with this board or not.

giuliogamba, 2019-04-02:
Which board are you referring to?

praveshgoyal, 2019-04-03:
@giuliogamba ,
 Zedboard or Zybo board

giuliogamba, 2019-04-05:
As previously answered in another issue (#10) hardware compatibility with other ZYNQ is not a problem, while the driver might need changes, unless you run the PYNQ framework on your zedboard (as done [here](https://superuser.blog/pynq-linux-on-zedboard/))

BrunoOliveira8, 2019-05-15:
It is possible to change some parameters and run on a mpsoc zu4EV?

giuliogamba, 2019-05-15:
@BrunoOliveira8 It is possible to port the HLS design to a different mpsoc device. I would suggest to start  the porting from the Ultra96 flow (ZU3). Which board are you trying to port to?

BrunoOliveira8, 2019-05-15:
Thank you for the answer. 
I start by analyse the .tcl files in the /script/ultra96 directory, but i am not sure the ports that i have to change. I want to port to a ZU4 board for my master dissertation.

giuliogamba, 2019-05-16:
Are you talking about ZCU104?

BrunoOliveira8, 2019-05-20:
Hello,
I am using this custom board: https://www.sundance.technology/system-on-modules-som/som-modules/teo820-zu4ev/
Any suggestion?

gsr-ee, 2019-07-20:
I successfully run the BNN on a Zedboard running on the PYNQ v2.3 with minor changes.