Xilinx/BNN-PYNQ #23: Problem about hardware design rebuilt

younglee1, 2018-03-05:
Hi!
I'm rebuilding the hardware design following the guide. But I've met some problem. 
[image]

As you can see, the problem occurs when running the **hardware synthesis**. It seems that the argument "-compile" doesn't work for the "csim_design" command. I tried to delete the argument, but it still fails with compilation error.
Could you please help me to solve the problem?

giuliogamba, 2018-03-05:
Hi, 

unfortunately to run the csim with C++11 requirements you will need to upgrade to Vivado 2017.1 or newer. 
You can still re-execute the HW rebuild flow by commenting out the the csim instruction in the hls-syn.tcl file.

younglee1, 2018-03-05:
Thanks, I'll try it.

closed 2018-03-06 by younglee1