Xilinx/BNN-PYNQ #24: Utilization Estimates of Synthesis

younglee1, 2018-03-06:
Hi，
Sorry for bothering again...But I've met another confusing problem.
Here is the Utilization of my synthesis result, which is much higher than yours, especially the LUT utilization. I wonder how this happens.  
My environment is Vivado 2017.1, Ubuntu 16.04.
[image]
Looking forward to your reply.

giuliogamba, 2018-03-06:
This is the Vivado HLS Estimate usage, that sometimes can be conservative. 
Vivado will perform further optimizations and you should see a smaller hardware usage.

younglee1, 2018-03-08:
Hi，
Since the synthesis result overuses the LUTs，I'm wondering can I accelerate part of BNN with FPGA？ Say pack convolutional layers and pooling layers as an IP, and leave FCLayers to
ARM core? Is that possible? If so, which files should I modify(top.cpp and ?)? And what should be paid attention to?

ps. I'm a total beginner to FPGA with no background knowledge. This project and your fast, kindly reply help me a lot. Thanks so much!

giuliogamba, 2018-03-08:
Hi, 

are you replicating the neural network that is included in this repository? If yes, the whole network fits in the device. 
As stated in the previous reply, the report you attached is the one generated by Vivado HLS, that sometimes overestimates the resources. You should run the complete flow including Vivado synthesis and place and route using the command `./make-hw.sh cnv-pynq pynq a`. Vivado is performing further optimizations that will allow the design to be correctly implemented.

younglee1, 2018-03-08:
OK，I'll retry it.
But I still want to know the answer to the question I mentioned above. What should I do if I just want to accelerate several layers of the neural network？

giuliogamba, 2018-03-08:
You just implement in the top.cpp the layers that you want to be offloaded in the PL, and execute the other layers in the PS.

younglee1, 2018-03-15:
Hi,
I want to run C/RTL Cosimulation in vivado HLS, but it always fails with "C++ compile error". In C simulation,  you choose clang as the compiler, but the default compile in Cosim is gcc. I have no idea how to change it. Could you please help me? How can I obtain the Cosim result?

giuliogamba, 2018-03-15:
Please upgrade your Vivado version. With versions > 2017.3 you can select the compiler to clang also when you run the RTL cosim

younglee1, 2018-03-15:
Thanks, I'll try it.

younglee1, 2018-03-17:
Hi,
I tried to run RTL cosim with vivado_hls 2017.4, and set clang as the compiler, but it still failed.
[image]
[image]

Why would this happen?

giuliogamba, 2018-03-20:
Hi, 

I don't know this problem, and I'm not able to reproduce it locally. I would assume there is a problem with your installation, since it's not able to find the clang-3.9 executable that should be installed with Vivado.

xuebao1942, 2018-07-17:
I have met the same problem as you. Have you solved it? @younglee1

giuliogamba, 2019-01-25:
Closed due to inactivity

closed 2019-01-25 by giuliogamba