Xilinx/BNN-PYNQ #26: Cifar10 cannot be launched in H/W after capturing Video through HDMI-in

shinya321, 2018-03-28:
Hi,
Now I am trying to **launch Cifar10 in H/W after capturing Video through HDMI-in.**
Software flow I wanna do is as below:

1. Capture a image through HDMI-in
2. Classify the image by CONV(Cifar-10)
3. Output the image with classified result through HDMI-out

# 1
At first when I tried classification for a local JPEG image before configuring HDMI-I/O, everything was fine like this:

```
>>> import bnn
>>> classifier = bnn.CnvClassifier('cifar10') # Launched in H/W
Setting network weights and thresholds in accelerator...
>>> from PIL import Image
>>> im = Image.open('./img/cifar10/car.jpg')
>>> class_out = classifier.classify_image(im)
Packing and interleaving CIFAR-10 inputs...
Running prebuilt CIFAR-10 test for 1 images...
Inference took 1584 microseconds, 1584 usec per image
Classification rate: 631.313 images per second
Inference took 1584.00 microseconds
Classification rate: 631.31 images per second
>>>
```

# 2
Second, I configured  Overlay for HDMI-i/o, and again I tried classification for a image captured by camera through HDMI-in. But not work, freeze in the following state:
```
>>> from pynq import Overlay
>>> from pynq.lib.video import *
>>> base = Overlay("base.bit")
>>> base.download()
>>> hdmi_in = base.video.hdmi_in
>>> hdmi_out = base.video.hdmi_out
>>> hdmi_in.configure(PIXEL_RGB)
<contextlib._GeneratorContextManager object at 0x2fc232b0>
>>> hdmi_out.configure(hdmi_in.mode, PIXEL_RGB)
<contextlib._GeneratorContextManager object at 0x2fc234f0>
>>> hdmi_in.start()
<contextlib._GeneratorContextManager object at 0x33d14110>
>>> hdmi_out.start()
<contextlib._GeneratorContextManager object at 0x33ee5790>
>>> inframe = hdmi_in.readframe()
>>> im = Image.fromarray(inframe)
>>> class_out = classifier.classify_image(im)
Packing and interleaving CIFAR-10 inputs...
Running prebuilt CIFAR-10 test for 1 images...

```

Where's wrong ?
Sorry to be poor English, it would be greatly appreciated if you could explain the details.

Best regards

giuliogamba, 2018-03-28:
Hi, 

the `classifier = bnn.CnvClassifier('cifar10')` instantiation actually download the overlay in the Programmable Logic featuring the hardware accelerator of the CNV BNN and download the trained weigths. 
If you download another bitstream after having instantiated the the classifier, when you run the classification the desired HW accelerator is not available. I suggest you to first download the base overlay, capturing the HDMI frame and then instantiate the CNV classifier. 

Hope this help

shinya321, 2018-03-28:
Thank you so much for replying me very quickly!!
As you told, I tried as below:

```
# 1. Configure HDMI-I/O
from pynq import Overlay
from pynq.lib.video import *
base = Overlay("base.bit")
base.download()
hdmi_in = base.video.hdmi_in
hdmi_out = base.video.hdmi_out
hdmi_in.configure(PIXEL_RGB)
hdmi_out.configure(hdmi_in.mode, PIXEL_RGB)
hdmi_in.start()
hdmi_out.start()
inframe = hdmi_in.readframe()

# 2. CNV instantiation and classification
from PIL import Image
import bnn
classifier = bnn.CnvClassifier('cifar10') # Launched in H/W
im = Image.fromarray(inframe)
class_out = classifier.classify_image(im)
```

-> everything was fine!
```
Packing and interleaving CIFAR-10 inputs...
Running prebuilt CIFAR-10 test for 1 images...
Inference took 1583 microseconds, 1583 usec per image
Classification rate: 631.712 images per second
Inference took 1583.00 microseconds
Classification rate: 631.71 images per second
```

And next, I tried to capture images from camera and classify each frame.
Here is code:
```
# ------------------------------------------
# 1. Configure HDMI-I/O
from pynq import Overlay
from pynq.lib.video import *
from PIL import Image
import bnn

base = Overlay("base.bit")
base.download()
hdmi_in = base.video.hdmi_in
hdmi_out = base.video.hdmi_out
hdmi_in.configure(PIXEL_RGB)
hdmi_out.configure(hdmi_in.mode, PIXEL_RGB)
hdmi_in.start()
hdmi_out.start()

# ------------------------------------------
# 2. CNV instantiation and classification

numframes = 20
for _ in range(numframes):
    inframe = hdmi_in.readframe()
    classifier = bnn.CnvClassifier('cifar10') # Launched in H/W
    im = Image.fromarray(inframe)
    class_out = classifier.classify_image(im)
    print(classifier.class_name(class_out))
    hdmi_out.writeframe(inframe)
    inframe.freebuffer()

hdmi_in.close()
hdmi_out.close()
```

But still freezed in the same state as I mentioned...
Anything's wrong??
I am so sorry to bother you again...

giuliogamba, 2018-03-28:
The CNV overlay does not have HDMI in or out, only the base overlay does. 
Every time you use HDMI, the base overlay has to be downloaded, and every time you use CNV the CNV overlay has to be downloaded. 
Another option is actually to integrate the CNV ip core within the base overlay, if you have experience in Vivado Block Design you can generate your own overlay with both CNV and HDMI.

shinya321, 2018-03-28:
Thanks again for your great reply.
I totally understand that CNV and HDMI Overlay provided in this git are independent and we cannot use them at the same time as a default.
I am just a software engineer and not familiar with Vivado...
Anyway thank you so much for telling me a lot!!

closed 2018-03-28 by shinya321