Xilinx/BNN-PYNQ #27: Inference result different after training the mnist dataset again

eerobert, 2018-04-10:
I can make-hw.sh with csim produce the correct inference result from git clone the repo.

However, when I go thru the training process on mnist (which result in 1.6% test error) and use the weights and threshold bin file to re-run the csim again. the inference result mismatch.

i wonder what step i am missing? anyone sucessfully re-run the bnn-pynq from training to inference and csim?

giuliogamba, 2018-04-10:
Hi, 

it's possible (not likely) that the single image we do use in csim is not classified correctly with the new weigths you trained. You can actually try with a different image by modifying the arguments passed to the csim (in the gui or in the make-hw.sh script) and check if, with a different image, if you get the expected results.

eerobert, 2018-04-11:
Hi,

Thanks for your reply. Somehow I cleanup the whole directory, rebuild everything and it works now

However, I now face another problem of changing the SIMD and PE number of the HLS.
Again, after changing the SIMD and PE number, the CSIM return the wrong class value.

What might be the problem? And how can we debug the issue?