Xilinx/BNN-PYNQ #29: Source code for low level driver

eerobert, 2018-04-23:
Hi,

I am porting the LFC to a smaller device and found that it cannot fits in.
So I modified the whole thing to SFC and it's already working on HLS CSIM.

Now I have a problem on loading the *.so and would like to seek for your help.
It would be very nice if you can at least release the API definition of those function in *.so

Or if possible source code will definitely good for further customization.

giuliogamba, 2018-04-23:
Hi, 

the source code for the generation of the .so is open-sourced as well [here ](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/lfc-pynq/sw/main_python.cpp). 
You can execute the command `make-sw.sh` script on the PYNQ board and compile it as explained [here](https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/src/network).
The important thing is that, at compilation time, the `main_python.cpp` is linking the `config.h` to get information about layer dimensions. 
Hope this help.

eerobert, 2018-04-24:
Hi,

OO.. thanks. I miss that point. 

Thanks so much.

eerobert, 2018-04-26:
Hi,

I tried to compiled the LFC PYTHON SW version, but the SW inference fail. 
It seems like the recompiled python_sw-lfc-pynq.so is different from the bundled so.
Would you mind provide me more instruction on this issue?

giuliogamba, 2018-04-26:
The .so might differ since it depends on the Vivado HLS library your compiling against. I've just re-compiled locally linking library from HLS 2017.4 and it works on my side, so I am not able to reproduce. 
Which version of the HLS library are you using?

eerobert, 2018-04-26:
Thanks for your prompt reply.

I am now able to make python_sw working. it seems like the on board g++ is not the same arm-linux-gnueabihf-g++ 2017.4.

But I still having problem in running python_hw. 

I wish to know why the "thePlatform" under foldedmv-offload is using the class donutdriver instead of xlnkdriver. The program throw an exception in the line foldedmv-offload.cpp:398 as the donutdriver return 0 on allocAccelBuffer.

giuliogamba, 2018-04-26:
Hi, 

are you running the executable as root?

eerobert, 2018-04-26:
Hi, 

I am using Xilinx SDK to step debug the executable. 

After reviewing the code, I wonder why the platform-xlnk.cpp:63 cannot override the donutdriver virtual allocAccelBuffer with xlnkdriver allocAccelBuffer.

eerobert, 2018-05-03:
Hi, 

After rebuild everything, it works. Thanks

closed 2018-05-03 by eerobert