Xilinx/BNN-PYNQ #30: Which version of Vivado HLS was used to rebuild the whole design?

eerobert, 2018-05-03:
Hi,

I am still working on this project and now I face another problem regarding Vivado HLS generated IP.

It seems like the generated IP get hang after writing ap_start. (no ap_done)

I would like to know which version of Vivado tools is suggested to re-run everything of this project.

giuliogamba, 2018-05-03:
The version used is 2017.4

eerobert, 2018-05-07:
Hi,

It is strange that I am using 2017.4 too, but the generated HLS IP is not working.

giuliogamba, 2018-05-08:
Which of the 2 networks in this repo is not working?

eerobert, 2018-05-09:
Hi,

Finally I make it working. 

My goal is to port BNN to a smaller device, hence I regenerate with SFC
And find that the most problematic module is the width conversion module.
In my case, the padded input bytes should be 800B. 
So in 64bit bus, it should read 13 words.
But in internal NN 32bit bus, it should write 25 words.
And the generated IP hang at this point.

closed 2018-07-10 by giuliogamba