Xilinx/BNN-PYNQ #31: After Binary Weights Files Generated

Alex-toto, 2018-05-26:
Hi, I am training the CNV network with cifar10 dataset. I got 'binparam-cnv-pynq' folder after using use 'python cifar10-gen-binary-weights.py', then you said place the resultant folder into the XILINX_BNN_ROOT/data directory on the Pynq device. Are you mean put the .bin files on the PYNQ board? 
I find there is no .bin files under /home/xilinx/jupyter_notebooks/bnn/data . Do I put the resultant folder in this folder? Where is the original weights for cifar10?

giuliogamba, 2018-05-28:
Hi, 

sorry, there's a typo in the README. The .bin files have to be moved under the `params` folder. If the bnn-pynq package is already installed, you should move the folder under `/opt/python3.6/lib/python3.6/site-packages/bnn/params` who contains all the trained parameters.

closed 2018-07-10 by giuliogamba

jdwo, 2019-02-01:
Hello,

it appears that on my PYNQ the directory `/opt/python3.6/lib/python3.6/site-packages/bnn/params` does not exist. In contrast, the actual path to the weights is `/usr/local/lib/python3.6/dist-packages/bnn/params`. 
Did this change with the latest updates?

Kind regards

giuliogamba, 2019-02-01:
Yes, it changed with PYNQ 2.3