Xilinx/BNN-PYNQ #34: Misc questions

mohdumar644, 2018-07-10:
1. I noticed an accuracy drop of upto 7% between training and inference when porting a 62 class dataset onto the LFC model. What could be the possible reasons for this?

2. Does the ConvLayer support padding/ 'same' convolution? (I believe not)

3. Does the MaxPooling Layer support odd dimension input images? (Again, I believe it does not).

4. Are there any rules to follow when inserting additional Conv layers in the CNV model? I seem to get a constant inferred class for all my inputs in such a model on two different datasets.

vonhachtaugust, 2018-07-30:
According to the paper number 2 is answered "Although not required by any of the networks described
in this work,  the SWU also pads the images  if  necessary." (last part of section 4.3.2)

Using the sliding window unit with equal input and output height and width gives me reasonable results but I have no definite answer by result availble, maybe I will produce some shortly. You just have to trust the authors on this one.

giuliogamba, 2018-07-30:
In this repository we have a limited set of the HLS building blocks using in FINN. For a more generic set you can refer to the FINN release [here](https://github.com/Xilinx/FINN/blob/master/FINN/backend/fpga/hls/).

closed 2018-10-10 by giuliogamba