Xilinx/BNN-PYNQ #42: New model HW layer acceleration

reshkor, 2018-09-03:
Hello,
I have added some extra layers on a CIFAR-10. I managed to make a successful simulation of the model and execute software inference on  PYNQ board. However if I try to run HW acceleration the results for all classes are 1. I guess the weights are not properly loaded on FPGA or some configuration in main_python.cpp is wrong (in OFFLOAD sections as software inference runs just fine).
Could you suggest how to fix this issue?

giuliogamba, 2018-09-03:
Hi, 
without having a look at the code it's very difficult to debug it. If you fork the repo and push your code in the forked repository, maybe I can have a look at it.

reshkor, 2018-09-03:
Thank you for your fast reply.
I have created a fork and pushed to https://github.com/reshkor/BNN-PYNQ (the time on PYNQ is wrong, so don't mid that commit was done 12 days ago).
The new network is called cifar10_v5 and that one actually has only 3 convolution layers.

Changyiyu, 2019-11-29:
@reshkor Hi, I referenced the link of [,](https://github.com/reshkor/BNN-PYNQ) and I don’t see the settings for these parameters in "cifar10-gen-binary-weights.py".
`#topology of convolutional layers (only for config.h defines)
    ifm       = [32, 30,  14,  12,   5,   3]
    ofm       = [30, 28,  12,  10,   3,   1]   
    ifm_ch    = [ 3, 64,  64, 128, 128, 256]
    ofm_ch    = [64, 64, 128, 128, 256, 256]   
    filterDim = [ 3,  3,   3,   3,   3,   3]

    WeightsPrecisions_fractional =    [0 , 0 , 0 , 0 , 0 , 0 , 0, 0,  0]
    ActivationPrecisions_fractional = [0 , 0 , 0 , 0 , 0 , 0 , 0, 0,  0]
    InputPrecisions_fractional =      [7 , 0 , 0 , 0 , 0 , 0 , 0, 0,  0]
    WeightsPrecisions_integer =       [1 , 1 , 1 , 1 , 1 , 1 , 1, 1,  1]
    ActivationPrecisions_integer =    [1 , 1 , 1 , 1 , 1 , 1 , 1, 1,  1]
    InputPrecisions_integer =         [1 , 1 , 1 , 1 , 1 , 1 , 1, 1,  1]

    classes = ['Airplane', 'Automobile', 'Bird', 'Cat', 'Deer', 'Dog', 'Frog', 'Horse', 'Ship', 'Truck']
`
Can these be omitted? Because I'm worried about changing the size of the input image.