Xilinx/BNN-PYNQ #43: Could you please tell me how FINN works to transfer Python codes to C++?

ThomasZhang50, 2018-09-04:
Dear Giuliogamba,
hello!
I am working with neural network on FPGA. And I'm very interested in your project. But could you please tell me how can you generate the "top.cpp"? Did you write some python codes and use "setup.py" to transfer it?
I am a beginner please excuse my easy question.
Sincerely,
Thomas Zhang
4. Sep

HPC-Ken, 2018-09-05:
Hi Thomas,

The top.cpp is generated from the FINN project https://github.com/Xilinx/FINN/.

Can you define what you mean by "transfer"?

ThomasZhang50, 2018-09-05:
Hello, Giuliogamba,
thanks for your Email!
I want to ask if you compose the python codes (for example "mnist.py" and
"setup.py") at first, then use for example cython to "transfer" the
"mnist.py" to "top.cpp"?
Now I've composed a neural network model in python. But I dont know how to
get VHDL codes from python codes. Do you mean FINN can transfer python
codes to VHDL?
Sincerly
Thonas Zhang
5. Sep

closed 2019-12-06 by giuliogamba