Xilinx/BNN-PYNQ #45: how to launch CO-Simulation in this Project

pochen125, 2018-09-21:
I have problem to launch Co-simulation .

Could you teach me. will I have any other configure need to set up?

giuliogamba, 2018-09-24:
Can you please elaborate on what problem do you have? Which version of Vivado tools are you using?

giuliogamba, 2018-09-24:
If you want to execute co-simulation, you can add a line to the tcl script of the HLS synthesis [here](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/hls-syn.tcl), after the `csynth_design` has been executed. The instruction should be: 
`cosim_design -argv "$directory_params $test_image 10 $expected_result" -compiler clang` 
added after line 78 and before the exit. Tested with Vivado HLS 2017.4.

giuliogamba, 2019-01-17:
Closing, solution provided

closed 2019-01-17 by giuliogamba