Xilinx/BNN-PYNQ #48: AttributeError: module 'bnn' has no attribute 'NETWORK_LFCW1A1'

weikaijie1994, 2018-10-05:
Excuse me, the project you have posted on Github named BNN-PYNQ. When I tried to execute the code you have posted. There is an attributeError which reminds me that the module 'bnn' has no attribute ‘NETWORK_LFCW1A1’. 
I have check the file named BNNs.py on the GitHub. There is actually an attribute named NETWORK_LFCW1A1. However, I have not any idea where it has been installed on the PYNQ. 
Would you like to give me any advice to solve this problem?

giuliogamba, 2018-10-05:
Have you installed the latest release of BNN-PYNQ on the latest PYNQ image (2.3)? 
If you had previously installed the BNN package, you should run the `pip `install command with the flag `--upgrade` in order to be sure the new version is properly installed.

weikaijie1994, 2018-10-05:
Actually, I have thought I have installed the latest version of PYNQ. When I run the command of "sudo pip3 install git+https://github.com/Xilinx/BNN-PYNQ.git (on PYNQ v2.3)" There comes error like 

"Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-ryq51liqb/"

Thus, I tried to applied the other command and it seems to have worked well. Although it reminds that 

"You are using pip version 9.0.1, however version 18.0 is available.
you should consider upgrading via the 'pip install --upgrade pip' command."

giuliogamba, 2018-10-05:
I suggest you to start clean, re-flash your SD-card with the latest PYNQ image following this [guide](https://pynq.readthedocs.io/en/latest/getting_started/pynq_image.html) and make a fresh install of the package.

weikaijie1994, 2018-10-05:
OK, I have fixed it through reinstall the PYNQ system. Thanks a lot.

closed 2018-10-08 by giuliogamba