Xilinx/BNN-PYNQ #50: executing bnn jupyter CnvClassifier:  libcma.so: cannot open shared object file

Urjitmodhia, 2018-10-08:
when running hw_classifier = bnn.CnvClassifier(bnn.NETWORK_CNVW1A1,'cifar10',bnn.RUNTIME_HW)

OSError: cannot load library /usr/local/lib/python3.4/dist-packages/bnn/libraries/pynqZ1-Z2/python_hw-cnvW1A1-pynqZ1-Z2.so: libcma.so: cannot open shared object file: No such file or directory

what to do?

giuliogamba, 2018-10-08:
Please use the latest PYNQ Image (2.3) available [here ](http://www.pynq.io/board.html).

closed 2018-10-30 by giuliogamba

Icarus0519, 2019-03-02:
> when running hw_classifier = bnn.CnvClassifier(bnn.NETWORK_CNVW1A1,'cifar10',bnn.RUNTIME_HW)
> OSError: cannot load library /usr/local/lib/python3.4/dist-packages/bnn/libraries/pynqZ1-Z2/python_hw-cnvW1A1-pynqZ1-Z2.so: libcma.so: cannot open shared object file: No such file or directory
> what to do?

Hi,Urjitmodhia,

I met the same issue,cant find the "python_hw-cnvW1A1-pynqZ1-Z2.so" file,Have you solved the issue?By using the image V2.3? I dont have a card reader at present,so i would like to ask if you have used other methods to solve this problem?looking forward to your reply.

kind regards.
icarus.