Xilinx/BNN-PYNQ #51: why do I execute this code at a slow speed(pynq-z2 and pynq2.3 images)?

YNZH, 2018-10-09:
this code is included in http://192.168.2.99:9090/notebooks/bnn/CNV-BNN_Cifar10.ipynb in pynq-z2(pynq2.3)。
Concretely，in my board:
[image]

Inference took 1582.00 microseconds
Classification rate: 632.11 images per second
   [CLASS]    [RANKING]
  Airplane       256
Automobile       379
      Bird       165
       Cat       169
      Deer       163
       Dog       244
      Frog       249
     Horse       244

      Ship       267
     Truck       268

but in this Repositories ,I see that:
[image]
Inference took 527.00 microseconds
Classification rate: 1897.53 images per second
   [CLASS]    [RANKING]
  Airplane       256
Automobile       379
      Bird       165
       Cat       169
      Deer       163
       Dog       244
      Frog       249
     Horse       244
      Ship       267
     Truck       268

as u see, the speed in my board  is three times slower.

giuliogamba, 2018-10-09:
The notebook screenshot uploaded have been executed on the Ultra96 board, which clocks the Neural Network accelerator at 300MHz, while you're executing on PYNQ-Z2 which is clocked at 100 MHz, thus the difference in execution time.

YNZH, 2018-10-09:
oh..thank u

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Xilinx/BNN-PYNQ/issues/51#issuecomment-428144863",
"url": "https://github.com/Xilinx/BNN-PYNQ/issues/51#issuecomment-428144863",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [Xilinx/BNN-PYNQ] why do I execute this code at a slow speed(pynq-z2 and pynq2.3 images)?  (#51)",
"sections": [
{
"text": "",
"activityTitle": "**giuliogamba**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@giuliogamba",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"Xilinx/BNN-PYNQ\",\n\"issueId\": 51,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"Xilinx/BNN-PYNQ\",\n\"issueId\": 51\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/Xilinx/BNN-PYNQ/issues/51#issuecomment-428144863"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 390795019\n}"
}
],
"themeColor": "26292E"
}
]

closed 2018-10-09 by giuliogamba