Xilinx/BNN-PYNQ #52: After training my own datasets and generating the binary weights

sxw228, 2018-10-10:
After training my own datasets and generating the binary weights, i get binparam-cnv-pynq .
How can i use these param in my PYNQ? My data was trained with lfc form. Do i need to rename the folder 
as lfcW1A1 and replace the former one in bnn\params\mnist?  If not , what should i do?

reshkor, 2018-10-11:
Hello,

TL;DR: rename your folder to lfcW1A1 and place it in bnn\params\myset, where myset can be any name for your network.

I have trained my dataset on gtrsb and got it running.
After generating the binary weights I got folder _binparam-cnvW1A1-pynq-gtrsb_, renamed it to _cnvW1A1_ and placed it in _bnn\params\myset_ (_myset_ is a folder I created). When I ran notebook (CNV-BNN_Cifar10.ipynb) `print(bnn.available_params(bnn.NETWORK_CNVW1A1))` showed _myset_ as an available network.
NOTE: It worked on the latest BNN-PYNQ release and on PYNQ v2.3 image.

Hope it helps.

closed 2018-10-13 by sxw228

FrancoCapraro, 2018-10-30:
> After training my own datasets and generating the binary weights, i get binparam-lfc-pynq .
> How can i use these param in my PYNQ? My data was trained with lfc form. Do i need to rename the folder
> as lfcW1A1 and replace the former one in bnn\params\mnist? If not , what should i do?

How you made your own datasets ? because i saw that you need files like this:
t10k-images-idx3-ubyte
train-images-idx3-ubyte
t10k-labels-idx1-ubyte
train-labels-idx1-ubyte