Xilinx/BNN-PYNQ #53: some questions about Hardware design rebuild

sxw228, 2018-10-15:
HI!
In order to intergrate the BNN ip core with my base overlay, i try to rebuild the hardware designs ,following the instructions.  After the ./make-hw.sh cnvW1A1 pynqZ1-Z2 a is finshed, i open the vivado project and there are some problems.

WARNING: [IP_Flow 19-3664] IP 'procsys_BlackBoxJam_0_0' generated file not found '/home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.srcs/sources_1/bd/procsys/ip/procsys_BlackBoxJam_0_0/procsys_BlackBoxJam_0_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'procsys_BlackBoxJam_0_0' generated file not found '/home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.srcs/sources_1/bd/procsys/ip/procsys_BlackBoxJam_0_0/procsys_BlackBoxJam_0_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'procsys_BlackBoxJam_0_0' generated file not found '/home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.srcs/sources_1/bd/procsys/ip/procsys_BlackBoxJam_0_0/procsys_BlackBoxJam_0_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'procsys_BlackBoxJam_0_0' generated file not found '/home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.srcs/sources_1/bd/procsys/ip/procsys_BlackBoxJam_0_0/procsys_BlackBoxJam_0_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'procsys_BlackBoxJam_0_0' generated file not found '/home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.srcs/sources_1/bd/procsys/ip/procsys_BlackBoxJam_0_0/procsys_BlackBoxJam_0_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [BD 41-1663] The design 'procsys.bd' cannot be modified due to following reason(s):
* Block design BD file is read-only. Please check file-permissions. 
* Project is read-only. Please run 'save_project_as' before making any change to the bd-design. 
* Block design BXML file is read-only. Please check file-permissions. 

WARNING: [BD 41-1661] One or more IPs have been locked in the design 'procsys.bd'. Please run report_ip_status for more details and recommendations on how to fix this issue.
List of locked IPs:
procsys_auto_pc_0
procsys_BlackBoxJam_0_0
procsys_ps7_0
procsys_rst_ps7_100M_0
procsys_ps7_axi_periph_0
procsys_axi_mem_intercon_0
procsys_auto_pc_1

It says that procsys_BlackBoxjam_0_0_sunth_1 Failed
So what's wrong. How can i fix this problem?

giuliogamba, 2018-10-15:
Which Vivado version are you using?

giuliogamba, 2018-10-15:
It seems the HLS IP has not been generated. Can you try first to run the `make-hw.sh` script with the flag `-h` to launch the HLS synthesis and then, after it finishes, execute the script with the flag `-b`?

sxw228, 2018-10-15:
After HLS synthesis, i got the following:

INFO: [Common 17-206] Exiting vivado_hls at Mon Oct 15 06:42:46 2018...
cat: /home/sxw228/BNN_PYNQ/bnn/src//network/output/hls-syn/cnvW1A1-pynqZ1-Z2/sol1/syn/report/BlackBoxJam_csynth.rpt: No such file or directory
HLS synthesis complete
HLS-generated IP is at /home/sxw228/BNN_PYNQ/bnn/src//network/output/hls-syn/cnvW1A1-pynqZ1-Z2/sol1/impl/ip
Done!

sxw228, 2018-10-15:
[image]
cp: cannot stat '/home/sxw228/BNN_PYNQ/bnn/src//network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper.bit': No such file or directory
cat: /home/sxw228/BNN_PYNQ/bnn/src//network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper_timing_summary_routed.rpt: No such file or directory
cat: /home/sxw228/BNN_PYNQ/bnn/src//network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper_utilization_placed.rpt: No such file or directory
cat: /home/sxw228/BNN_PYNQ/bnn/src//network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper_utilization_placed.rpt: No such file or directory
cat: /home/sxw228/BNN_PYNQ/bnn/src//network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper_utilization_placed.rpt: No such file or directory
cat: /home/sxw228/BNN_PYNQ/bnn/src//network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper_utilization_placed.rpt: No such file or directory
Bitstream copied to /home/sxw228/BNN_PYNQ/bnn/src//network/output/bitstream/cnvW1A1-pynqZ1-Z2.bit
Done!

And this is the result of the make-hw.sh script with -b.

giuliogamba, 2018-10-15:
The problem is that HLS has not been launched. Are you sure you have Vivado HLS installed? Can you attach all the log of Vivado hls?

giuliogamba, 2018-10-15:
As you can see from the log, HLS encountered an error: `Error in linking the design`. Can you please erase the HLS output subfolder `BNN_PYNQ/bnn/src//network/output/hls-syn` before executing the `make-hw.sh` script with `-h` flag?

sxw228, 2018-10-15:
[vivado_hls.log](https://github.com/Xilinx/BNN-PYNQ/files/2479671/vivado_hls.log)
Here is the hls log.
I erased the output folder and rebuild again, but the problem still exists.

ARNING: [IP_Flow 19-3664] IP 'procsys_BlackBoxJam_0_0' generated file not found '/home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.srcs/sources_1/bd/procsys/ip/procsys_BlackBoxJam_0_0/procsys_BlackBoxJam_0_0.dcp'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'procsys_BlackBoxJam_0_0' generated file not found '/home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.srcs/sources_1/bd/procsys/ip/procsys_BlackBoxJam_0_0/procsys_BlackBoxJam_0_0_stub.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'procsys_BlackBoxJam_0_0' generated file not found '/home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.srcs/sources_1/bd/procsys/ip/procsys_BlackBoxJam_0_0/procsys_BlackBoxJam_0_0_stub.vhdl'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'procsys_BlackBoxJam_0_0' generated file not found '/home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.srcs/sources_1/bd/procsys/ip/procsys_BlackBoxJam_0_0/procsys_BlackBoxJam_0_0_sim_netlist.v'. Please regenerate to continue.
WARNING: [IP_Flow 19-3664] IP 'procsys_BlackBoxJam_0_0' generated file not found '/home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.srcs/sources_1/bd/procsys/ip/procsys_BlackBoxJam_0_0/procsys_BlackBoxJam_0_0_sim_netlist.vhdl'. Please regenerate to continue.
WARNING: [BD 41-1663] The design 'procsys.bd' cannot be modified due to following reason(s):
* Block design BD file is read-only. Please check file-permissions. 
* Project is read-only. Please run 'save_project_as' before making any change to the bd-design. 
* Block design BXML file is read-only. Please check file-permissions. 

WARNING: [BD 41-1661] One or more IPs have been locked in the design 'procsys.bd'. Please run report_ip_status for more details and recommendations on how to fix this issue.
List of locked IPs:
procsys_auto_pc_0
procsys_BlackBoxJam_0_0
procsys_ps7_0
procsys_rst_ps7_100M_0
procsys_ps7_axi_periph_0
procsys_axi_mem_intercon_0
procsys_auto_pc_1

sxw228, 2018-10-16:
INFO: [Common 17-206] Exiting Vivado at Tue Oct 16 03:10:23 2018...
cp: cannot stat '/home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper.bit': No such file or directory
cat: /home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper_timing_summary_routed.rpt: No such file or directory
cat: /home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper_utilization_placed.rpt: No such file or directory
cat: /home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper_utilization_placed.rpt: No such file or directory
cat: /home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper_utilization_placed.rpt: No such file or directory
cat: /home/sxw228/BNN_PYNQ/bnn/src/network/output/vivado/cnvW1A1-pynqZ1-Z2/cnvW1A1-pynqZ1-Z2.runs/impl_1/procsys_wrapper_utilization_placed.rpt: No such file or directory
Bitstream copied to /home/sxw228/BNN_PYNQ/bnn/src/network/output/bitstream/cnvW1A1-pynqZ1-Z2.bit

when impl_1 finished, i get  errors  above.And the building is done.

I open the project, it turns out that procsys_BlackBoxjam_0_0_sunth_1 failed .
Can you give me some advice?

Farheen-github, 2020-02-17:
[image]
Can you please help me out : how did you resolve your issue as mentioned above.
I'm facing the similar one

giuliogamba, 2020-09-30:
Sorry, this comment must have slipped my attention. Do you experience the same problem with the scripts provided for PYNQ or Ultra96?

Farheen-github, 2020-10-25:
oh hy, I didn't receive the notification for this email
It has been resolved, thank you so much for consideration

closed 2020-10-27 by giuliogamba

lzh258369, 2021-06-01:
when I generate the bitstream, also meat the same problem, how do you solve this problem. Thank you.