Xilinx/BNN-PYNQ #56: Make my a training with my own images

FrancoCapraro, 2018-10-26:
Hi,  i have a pack of images and i want to **make the training of based on them.** 
I saw (as example) that the mnist.py (from BNN-PYNQ/bnn/src/training) i**mport the datasets from pylearn2** (from pylearn2.datasets.mnist import MNIST), inside of this i can see :
t10k-images-idx3-ubyte  
train-images-idx3-ubyte
t10k-labels-idx1-ubyte  
train-labels-idx1-ubyte

**So here is my question. How can i do my own "ubyte files" from my own pack of images to do the training of them.**

giuliogamba, 2018-10-30:
This is not a question related to this release, while a Machine learning question. 
Nonetheless, please check [here](https://github.com/gskielian/JPG-PNG-to-MNIST-NN-Format ), it should help you generating your own dataset.

FrancoCapraro, 2018-10-30:
> This is not a question related to this release, while a Machine learning question.
> Nonetheless, please check [here](https://github.com/gskielian/JPG-PNG-to-MNIST-NN-Format), it should help you generating your own dataset.

sorry for that and thank you for your help !

nickfraser, 2018-10-30:
Just a quick follow-up, you can also import your images as numpy arrays if you wish. If you print the dimensions of the arrays for [inputs](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/training/mnist.py#L113) and for [labels](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/training/mnist.py#L123) you should be able to see what's going on.

You would only need to replace those arrays with your preferred dataset. If everything is in the same order and the same size, everything should work.

closed 2018-10-30 by nickfraser