Xilinx/BNN-PYNQ #57: rebuild error

feilongwang, 2018-10-29:
Hi! When I used vivado 2018.2 to rebuild the repo, some errors are as followed:
root@wfl-ThinkPad-T440:/home/wfl/BNN-PYNQ/bnn/src/network# ./make-hw.sh cnvW1A1 pynqZ1-Z2 h
xilinx-tiny-cnn already cloned
Calling Vivado HLS for hardware synthesis...

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.2 (64-bit)
  **** SW Build 2258646 on Thu Jun 14 20:02:38 MDT 2018
  **** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source /opt/Xilinx/Vivado/2018.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/opt/Xilinx/Vivado/2018.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'root' on host 'wfl-ThinkPad-T440' (Linux_x86_64 version 4.15.0-29-generic) on Mon Oct 29 07:45:10 CST 2018
INFO: [HLS 200-10] On os Ubuntu 18.04.1 LTS
INFO: [HLS 200-10] In directory '/home/wfl/BNN-PYNQ/bnn/src/network/output/hls-syn'
HLS project: cnvW1A1-pynqZ1-Z2
HW source dir: /home/wfl/BNN-PYNQ/bnn/src/network/cnvW1A1/hw
BNN HLS library: /home/wfl/BNN-PYNQ/bnn/src/library/hls
INFO: [HLS 200-10] Opening project '/home/wfl/BNN-PYNQ/bnn/src/network/output/hls-syn/cnvW1A1-pynqZ1-Z2'.
INFO: [HLS 200-10] Adding design file '/home/wfl/BNN-PYNQ/bnn/src/network/cnvW1A1/hw/top.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/wfl/BNN-PYNQ/bnn/src/network/cnvW1A1/hw/../sw/main_python.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/wfl/BNN-PYNQ/bnn/src/library/host/foldedmv-offload.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/wfl/BNN-PYNQ/bnn/src/library/host/rawhls-offload.cpp' to the project
INFO: [HLS 200-10] Opening solution '/home/wfl/BNN-PYNQ/bnn/src/network/output/hls-syn/cnvW1A1-pynqZ1-Z2/sol1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg400-1'
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch CLANG as the compiler.
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Mon Oct 29 07:45:11 2018...
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
Error in Vivado_HLS

Please help ,Thank you!

feilongwang, 2018-10-31:
yes, I used vivado 2018.2, maybe I shouled reinstall it.

giuliogamba, 2019-01-17:
@feilongwang Have you solved this problem?

maunns, 2019-04-15:
@giuliogamba HI,

I have the same problem as follows:-

$ ./make-hw.sh cnvW1A2 ultra96 a
xilinx-tiny-cnn already cloned
DEBUGGING #1
Calling Vivado HLS for hardware synthesis...
DEBUGGING #2
vivado_hls -f E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/hls-syn.tcl -tclargs cnvW1A2-ultra96 E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/cnvW1A2/hw E:/masters/special-problem/BNN-PYNQ-master/bnn/src/../params/cifar10/cnvW1A2 E:/masters/special-problem/BNN-PYNQ-master/bnn/src/../../tests/Test_image/deer.bin 4 xczu3eg-sbva484-1-i 3

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.3 (64-bit)
  **** SW Build 2405991 on Thu Dec  6 23:38:27 MST 2018
  **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source E:/Vivado/2018.3/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running 'E:/Vivado/2018.3/bin/unwrapped/win64.o/vivado_hls.exe'
INFO: [HLS 200-10] For user 'Muhammad Shabbir' on host 'm-shabbir' (Windows NT_amd64 version 6.2) on Mon Apr 15 16:40:26 -0400 2019
INFO: [HLS 200-10] In directory 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/output/hls-syn'
HLS project: cnvW1A2-ultra96
HW source dir: E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/cnvW1A2/hw
BNN HLS library: E:/masters/special-problem/BNN-PYNQ-master/bnn/src/library/hls
INFO: [HLS 200-10] Opening project 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/output/hls-syn/cnvW1A2-ultra96'.
INFO: [HLS 200-10] Adding design file 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/cnvW1A2/hw/top.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/cnvW1A2/hw/../sw/main_python.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/library/host/foldedmv-offload.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/library/host/rawhls-offload.cpp' to the project
INFO: [HLS 200-10] Opening solution 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/output/hls-syn/cnvW1A2-ultra96/sol1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 3ns.
INFO: [HLS 200-10] Setting target device to 'xczu3eg-sbva484-1-i'
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch CLANG as the compiler.
make: 'csim.exe' is up to date.
Setting network weights and thresholds in accelerator...
@E Simulation failed.
ERROR: [SIM 211-100] CSim failed with errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Mon Apr 15 16:40:33 2019...
DEBUGGING #3
ERROR: [SIM 211-100] CSim failed with errors.
Error in Vivado_HLS

Could you help with it??