Xilinx/BNN-PYNQ #60: Training of WnAn

mohdumar644, 2018-11-12:
I noticed an update to the BNN-PYNQ library adding support for WnAn.

Can you please give a pointer as to what quantization scheme are you using, and what framework?

From the finnthesizer, it does not look like you are using DoReFa-Net like you used in QNN-MO-PYNQ.

nickfraser, 2018-11-12:
We're planning to release the training scripts by the end of this week - I hope this will answer all of your questions.

mohdumar644, 2018-11-23:
Can you link a research paper, if you have implemented that?

mohdumar644, 2019-01-23:
Thanks, but your paper did not define the quantization function exactly, and I could not decipher the function from the finnthesizer. Are there more any more hints?

mohdumar644, 2019-02-03:
What does the s0.5 and s0.25 imply in the given ``.npz`` names?

nickfraser, 2019-02-06:
There is actually no (relevant) meaning to those. It's related to the topological description of CNV & LFC - these values are simply hardcoded into the cnv/lfc.py files respectively.

Apologies again for the delay, we plan to upload the training scripts for these this week.

mohdumar644, 2019-02-07:
Thanks, I was able to train a 2 bit and a 4 bit network using the -.5,.5 and equidistant thresholds respectively.

closed 2019-02-07 by mohdumar644