Xilinx/BNN-PYNQ #62: Bad detection using LFC-BNN-MNIST-Webcam

FrancoCapraro, 2018-11-13:
Hi, im using LFC-BNN-MNIST-Webcam.ipynb to see how its work. I'm getting as response "0", when im capturing "1", so my question is: 
When I do a:
`jupyter notebook LFC-BNN_MNIST_Webcam.ipynb` 
how can i know if it was correctly programmed the FPGA? (with the lfcW1A1-ultra96.bit and lfcW1A1-ultra96.tcl)

maybe Im doing something wrong or i need make other step/s before the jupyter command

I do the  " sudo pip3 install git+https://github.com/Xilinx/BNN-PYNQ.git " obviously

giuliogamba, 2018-11-15:
Can you please elaborate your question? 
You can test the hardware correctness by executing this [notebook](https://github.com/Xilinx/BNN-PYNQ/blob/master/notebooks/LFC-QNN_MNIST.ipynb) which is testing with the MNIST testset.

FrancoCapraro, 2018-11-15:
Hi, i was testing the hardware using the LFC-BNN-MNIST-Webcam.ipynb but writing my own numbers trying to copy the numbers as MNIST but i got bad number decision.
My  questions are:
1) When you test one image taken from the webcam so you do the 9 Steps, but **if i want test another one , what do you recommend to do ? Close the notebook and reopen a new one ?**
( i have some problems in the part when you take the image with the webcam doing the second test)

2) If i export the ipynb to python, i got the same number decision "2" (misc.imsave('/home/xilinx/img_webcam_mnist.png', img_data) ) so, **if i want use the python code i need to change that image , right?**

giuliogamba, 2018-11-15:
It might be a problem of pre-processing. The number has to be properly centered within the image (at the output of point 6 of the notebook). 
If you want to re-run with a new image, you should restart from point 4 onwards. 
I don't understand the second question.

FrancoCapraro, 2018-11-15:
ok! nice.
About the second question: I want to use the "python LFC-BNN-MNIST-Webcam.py " action to classificate an image. To do that , i downloaded the python code generated for the Jupyter Notebook in File>Download>Python.
 **So, i need to change the image : /home/xilinx/img_webcam_mnist.png ?** Because when i do the "python"  i got the same response "2"**( i thought that is because that image is a "2" and im not changing it so the response will not change)**

giuliogamba, 2018-11-15:
If you provide the same input image to the same neural network you will get the same classification, being the implementation deterministic, independently if  it's through python or Jupyter notebook.

closed 2018-11-15 by giuliogamba