Xilinx/BNN-PYNQ #66: Using the Params of CNVW1A1

FrancoCapraro, 2018-12-03:
First of all if you dont know about this guide you can close this Issue.
Hi, im using this guide/tutorial from Xilinx:

https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841949/Zynq+UltraScale+MPSoC+Accelerated+Image+Classification+via+Binary+Neural+Network+TechTip

This Tutorial use a CNVW1A1 for Road Signs. I'm trying to change the weights and thres with another training, for example the params of BNN-PYNQ, but i saw some problems to use it.

The params in this proyect ( /aic/src_hw/bnn-pynq-mpsoc/params/road-signs )have a different format that Params used in the Git of BNN-PYNQ( https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/params/road-signs/cnvW1A1 ) so, How can i make a training for this Tutorial or reuse the Params in BNN-PYNQ ?

giuliogamba, 2018-12-06:
Hi, 

we do have support to generate weights for run-time updates, as shown in this repo, or to generate .h files to be used directly in the HLS code, thus fixing the values at compile time (in the bitstream). 
The feature is not officially supported in this repo and not shown, but if you're willing to check it yourself the code call is just commented out in the script to export [weights ](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/training/cifar10-gen-weights-W1A1.py) like in lines 106 and 166.

FrancoCapraro, 2018-12-06:
Okay thank you so much, i will try it.

closed 2018-12-06 by FrancoCapraro