Xilinx/BNN-PYNQ #70: Specify address for storage

leankemski, 2018-12-17:
Hi,

I am trying to specify an address for storage because I want to use external memory. My board is ZYNQ-Z1.  I want to know will it work if change the ADDR_REG_BASE to external address in platform-xlnk.cpp and rebuild haedware design? What else should I change? And how much memory is needed to store all the data? 

Please help, thank you.

giuliogamba, 2019-01-03:
Can you please clarify your goals? 
The value `ADDR_REG_BASE` is `platform-xlnk.cpp` has to match the address set-up in the address editor in Vivado. The weights and thresholds are read from files and passed to the hardware accelerator via AXI-Lite, thus having very small memory requirements (in this case, 1024 registres of 64 bits each).

giuliogamba, 2019-01-17:
@leankemski Does that answer your question?