Xilinx/BNN-PYNQ #71: Version of Linux to make trainings with GPU-Theano

FrancoCapraro, 2018-12-28:
Hi, im trying to follow the steps in the "README.md" file inside /src/training , but i tried with ubuntu 16.04 and 18.04 i got a broken linux every time that i want to use Theano with GPU, Specially when im installing the drivers for the GPU. (Hopelly im using virtualenv to try it)
 So, my questions are:
1) What version (or distribution) of Linux you recommend  to use it ?
2) In the README.md speak about some Drivers of cuDNN and Cuda that they are old right now, Did you try with new versions ?

for info:
GPU: GTX 1060 Max Q

FrancoCapraro, 2018-12-28:
When i do the reboot to check that the Nvidia driver is working by running:nvidia-smi
cant login again (loop login) so i dont know what im doing wrong

nickfraser, 2019-01-15:
Apologies for the delay.

Supporting Nvidia driver installation is beyond the scope of this project/repository. We recommend Ubuntu 16.04 and CUDA 8.0, because we have verified that these work with Theano/Lasagne. These drivers can be difficult to install on certain systems, I recommend you try all of the methods available [here](https://developer.nvidia.com/cuda-80-ga2-download-archive).

We plan to provide docker images which will make the environment easier to set up, but I cannot provide a timeline for this right now.

closed 2019-01-15 by nickfraser

FrancoCapraro, 2019-04-08:
> Apologies for the delay.
>
> Supporting Nvidia driver installation is beyond the scope of this project/repository. We recommend Ubuntu 16.04 and CUDA 8.0, because we have verified that these work with [...]

Any new info about the docker ? 
Best
FC

giuliogamba, 2019-04-08:
They're already available [here](https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/src/training/docker)