Xilinx/BNN-PYNQ #72: Problem about hardware design rebuilt, 'Vivado not found'

jihye0005, 2019-01-07:
Hi!
I installed Vivado2018.2. 
I'm trying to rebuild the hardware desing, but have a problem.

[image]

Help me please~.
Thanks a lot!

giuliogamba, 2019-01-07:
Possibly you installed Vivado but haven't sourced it. 
Please refer to pag.39 of this [document](https://www.xilinx.com/content/dam/xilinx/support/documentation/sw_manuals/xilinx2018_3/ug973-vivado-release-notes-install-license.pdf)

closed 2019-01-07 by giuliogamba

maik123456789, 2020-08-24:
HI! I have the same problem but I cannot solved, what am I doing wrong? 
[image]