Xilinx/BNN-PYNQ #76: Real-time object detection (streaming video) with BNN

houssem123456, 2019-02-07:
Hi Giuliogamba,
Actually, I'm interested in real-time object detection on pynq-z1 using the  BNN  project .
Indeed, I want to have as input a video on the HDMI IN of the card and I'd like to have as output (on the HDMI OUT) the detection of all the objects in this video.
Please, can you tell me if that is possible?
 if so please tell me what are the things that must be done to achieve that?

giuliogamba, 2019-02-18:
Hi, 
[this](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841949/Zynq+UltraScale+MPSoC+Accelerated+Image+Classification+via+Binary+Neural+Network+TechTip) wiki might be helpful for you. It shows the implementation of object detection on the Ultra96 or ZCU102 board, and it's based on the BNN IP available on this repo. 

Hope this helps.

closed 2019-02-18 by giuliogamba

houssem123456, 2019-03-01:
Thank you for your reply 
I appraciate it