Xilinx/BNN-PYNQ #79: Hi,

saraballeri, 2019-02-15:


giuliogamba, 2019-02-18:
We do use some convenience functions from tiny-cnn, like the load of input images. 
Additionally some layers have been added, like the OFFLOAD, is order to manage the hardware offload of the network.

closed 2019-02-18 by giuliogamba

saraballeri, 2019-02-19:
Hi,
thank for your response.
But in particular what is the aim of these two functions in the loop?
```
for(unsigned int i = 0; i < count; i++) {
    tiny_cnn::vec_t interleaved = interleaver.forward_propagation(imgs[i], 0);
    quantiseAndPack<inWidth, 1>(interleaved, &packedImages[i * psi], psi);
}
```
Is there a documentation about tiny-cnn other than github repo?

Thanks,

Sara

aquibjamal, 2021-03-08:
Hi,

I am trying to use the tiny_cnn library on ESP32 board.
There is a reboot when the code reaches the declaration of **interleaver** in the function **testPrebuiltCIFAR10_from_image()**.
I would be grateful for any support in understanding what is the use of **interleaver** and if there is any workaround to directly use the **imgs** vector if I don't want to use interleaved inputs for inference.

With regards
Aquib