Xilinx/BNN-PYNQ #81: Number of PEs and SIMDs in LFCW1A1

jdwo, 2019-02-18:
Hello,

I am trying to fully understand FINN. 

I am trying to figure out, if it's possible to change the number of PEs and SIMDs for the implementation of the LFCW1A1 to better understand the underlying structure. I want to test how the change would affect the throughput of the accelerator.

I trained the network on MNIST with the mnist.py-script and generated the parameters for the different values of PEs and SIMDs with the gen-weights-script. Then I changed the config-file and the parameters and tried simulation in VivadoHLS. It failed with all parameters I tried.

Or is this approach meaningless? If so, what would be a better approach?

Thank you for your help

Kind regards

giuliogamba, 2019-02-18:
When you change the PE and SIMD values, the python scripts generates both the hw-config file and the parameters .bin files. You should move the new `config.h` file in the hardware [folder](https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/src/network/cnvW1A1/hw) and move the generated parameters in the param [folder ](https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/params/cifar10/cnvW1A1). Is this what you have done? 
If yes, what king or error you see?

jdwo, 2019-02-18:
I moved both the new `config.h` as well as the new parameters in the param folder.

This error message showed up:
`while executing
"source .../BNN-PYNQ/bnn/src/network/output/hls-syn/lfcW1A1-pynqZ1-Z2/sol1/csim.tcl"
    invoked from within
"hls::main .../BNN-PYNQ/bnn/src/network/output/hls-syn/lfcW1A1-pynqZ1-Z2/sol1/csim.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"`

After repeating the steps for the hardware design rebuild it shows more information. The classification result is wrong. It classifies the test image as class 0 instead of 4. I am going to try different values now, to see if the problem remains.

closed 2019-02-18 by jdwo

Changyiyu, 2019-12-04:
@jdwo Hi, 
What do you base on when you change the value of PEs and SIMDs? I changed the pixel size 32x32 of cifar10 to 128x128, but I don’t understand whether the value of PEs and SIMDs. Can you give me direction or resources?