Xilinx/BNN-PYNQ #82: Generation of Shared Libraries

jdwo, 2019-02-19:
Hello,

I generated the bit and tcl-file for a value of PEs and SIMDs. Then, I changed the bnn.py-class on the PYNQ to feature a new network attribute, but now I have the problem that I am missing the Shared-Object-Libraries for that attribute.

I did this, so I can keep the unchanged bit- and tcl-files on the board for comparison. 

How can I create this library for the new attribute? Or to rephrase the question, how are those libraries created in the first place? I am new to CFFi, any advice would help.

Thank you for your help

Kind regards

giuliogamba, 2019-02-19:
I don't think I've understood your question, but the `make-sw.sh` available [here](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/make-sw.sh) is the script used to compile (on the board) the shared object linked from [bnn.py ](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/bnn.py) python class. 
In order to compile the `.so` on the board you need to move the include folder available in your Vivado HLS installation path on the board and set the env variable `VIVADOHLS_INCLUDE_PATH` to the absolute path.

jdwo, 2019-02-19:
Ok, I created the new .bit and .tcl-files (PE = [32, 32, 32, 16]) and put them in the `/usr/local/lib/python3.6/dist-packages/bnn/bitstreams`-directory. I renamed the existing `lfcW1A1-pynqZ1-Z2.*`-files, and named the new generated ones as  `lfcW1A1-pynqZ1-Z2.*`. I then changed the parameters in `.../params/mnist/lfcW1A1` to the ones, which I generated with the adapted `gen-weights-lfcW1A1.py`. 

If I try to start inference, the kernel dies.

I hope that I did not forget to share important information this time.

Thank you for your help.

giuliogamba, 2019-02-19:
Have you changed the interface of the top function in HLS?

jdwo, 2019-02-19:
No, I did not change the top function. This might be a stupid question, what do I need to change there?
I did recompile using `make-sw.sh`, but I  did indeed forget to change the `config.h`. I recompiled it now with the new `config.h` on the board.

giuliogamba, 2019-02-19:
Nothing to change in the top function. Does the new .so fixed your problem?

jdwo, 2019-02-19:
Ok, good. I thought maybe I missed something. 
I copied the new `lfcW1A1-pynqZ1-Z2` from `../bnn/src/network/output/sw` to `../bnn/libraries/pynqZ1-Z2/` and tried to do inference again, but the kernel still dies at the same step. So it did not fix the problem.

giuliogamba, 2019-02-19:
you should move the freshly compiled  .so under `/usr/local/lib/python3.6/dist-packages/bnn/libraries/pynqZ1-Z2/`

giuliogamba, 2019-02-19:
Ok, so if you launched the `make-hw.sh` script after having changed the params, that means that Vivado HLS csim has passed. 
Can you please check what error is raised by checking `/var/log/jupyter.log`?

jdwo, 2019-02-19:
this entry is made when the kernel dies:
```
... NotebookApp] Adapting to protocol v5.1 for kernel ff75a513-9aa5-426b-b30d-c46b1c751c0d
terminate called after throwing an isntance of 'char const*'
```

giuliogamba, 2019-02-26:
Have you tried with the latest update?

gsr-ee, 2019-06-30:
I'm having the same problem. I trained the network for a different precision, the weights and the config.h were generated. I also modified the top.cpp to adapt the network a different precision. I run successfully the Synthesis in the Vivado HLS and the bitstream generation in Vivado. I followed the steps to generate de shared libraries and I made it. But, when I load the shared libraries and the parameters by using the bnn.CnvClassifier, the kernel dies.

Any help would be useful.

giuliogamba, 2019-07-01:
@gsr-ee Have you recompiled the .so file for the host?

gsr-ee, 2019-07-04:
No, didn't do that.

I'll talk you through what I've done. Hope you can help me.

First, I've changed the precision of the layers and the precisions of the weight type following the instructions [here](https://github.com/Xilinx/BNN-PYNQ/issues/69#issuecomment-477162348). I used 1 bit for integer and 1 bit for fractional. Then, I trained the network with my own dataset and I generated the binary weights files and put them into a new folder in /bnn/params (I did the same on the cnvW2A2 and it worked well). I made use of the files in /bnn/src/network/cnvW2A2/ to generate de shared libraries for the new network, which I called cnvW11A11. I made a fork of the project [here](https://github.com/gsr-ee/BNN-PYNQ/). 
Last but not least, I tried to load the shared libraries and the parameters on a new notebook file by using the bnn.CnvClassifier e this error ocurred.

Maximilianxu, 2019-11-27:
@giuliogamba Hi, I am new to HLS. What does "make-sw.sh" do? Does it generate drivers? How can "g++" used in "make-sw.sh" generate drivers for FPGA?

giuliogamba, 2019-12-05:
@Maximilianxu when executed on the PYNQ board, the `make-sw.sh` compiles shared objects that are used from the [bnn python class](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/bnn.py). 
When executed in the `python_hw` mode it does generate the drivers communicating with the hardware accelerator.

closed 2020-09-30 by giuliogamba