Xilinx/BNN-PYNQ #85: Real-time object detection with BNN-PYNQ project

houssem123456, 2019-03-07:
Hi Giuliogamba,
Actually, I'm interested in real-time object detection on pynq-z1 using the BNN-PYNQ project.
Indeed, I want to have as input a video on the HDMI IN of the card and I'd like to have as output (on the HDMI OUT) the classification of the object present in this video.
So my question : 
Is it possible  to add an AXI-stream to the BlackBoxJam ? 
Best Regards.

paulr92v, 2019-03-25:
Hi @giuliogamba,

I have a basic video setup using the CNVw1a1 traffic sign network using pynq.
I'm reading the video frames from a usb webcam using the opencv (the read part is done in a separate thread), then the 640x480 frame is broken down into ~750 tiles(32x32 pixels), for each tile the classify method is called.
Currently the performance is quite slow: about 6fps, latency is about 1.5 seconds.
I would like to achieve at least 15fps and latency under ~500ms, ca you please advise if this is possible?
I tried to modify the vivado project and add a second BlackBox_jam IP but this is not possible since only one IP consumes 57% BRAM. My intention was to use 2 IPs instead of one in order to improve performance.

I have little to no experience in Vivado HLS design, but I have very good knowledge of HDLs, do you think it is possible to improve performance by writing the basic building blocks of the network in HDL instead of HLS?

Regards,
Paul

giuliogamba, 2019-04-02:
There are several things you can explore:
1. Increase the CNV classification speed by increasing the per-layer parallelism (SIMD and PE counts [here](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/training/cifar10-gen-weights-W1A1.py)). This can easily give you a 3x speed-up you need. 
2. Include the tiles generation from the VGA image in the hardware. This would decrease the image pre-processing time. 
Hope this helps.

closed 2019-04-02 by giuliogamba