Xilinx/BNN-PYNQ #87: Training my own dataset

Sayf-Khan, 2019-03-11:
Hello,
I do apologize as this is not a question to do with the repo itself but more general to do with machine learning but could you elaborate on the following:

CNV:
1 - Import training data and store in a Nx3x32x32 numpy array, with each value scaled from -1 to 1.
2 - Put each output label into one hot encoded format.
2 - Break the inputs and outputs up into training, validation and test sets.
3 - Replace the lines in cifar10.py which load the training, validation and test sets with your newly imported and formatted numpy arrays.

Kind regards,
Sayf Khan

nickfraser, 2019-03-12:
The instructions you have pasted in your issue seem pretty clear to me - which step are you having trouble with?

Sayf-Khan, 2019-03-12:
Hello,

Primarily it is step 1.
1 - Import training data and store in an Nx3x32x32 numpy array, with each value scaled from -1 to 1.

Looking at the CIFAR10.py file in pylearn2 folder and cifar10.py file in the training folder, I see that the data batches of that image set are loaded. 

I am confused on how to import the image set I have?
- Shall I replace the data_batches files and if so how

Kind regards,
Sayf Khan

nickfraser, 2019-03-12:
I would not suggest modifying any code in `pylearn2`. At the end of [line 126 in cifar10.py](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/training/cifar10.py#L126), `train_set.X` is a 4D numpy array (image_index x colour x height x width), where every entry in the array is between -1 and 1.

Some understanding of numpy is prerequisite knowledge to this work, if you need more information I would strongly recommend doing some numpy tutorials. In particular, any tutorials which explain how to import images into numpy arrays.

closed 2019-03-12 by nickfraser