Xilinx/BNN-PYNQ #88: Generating the bitstream after running command persisting after deleting hls-syn folder

jbennett3, 2019-03-12:
i get the following error after removing the hls-syn when trying to generate the bitstream
when entering: ./make-hw.sh cnvW2A2 pynqZ1-Z2 a 

ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Tue Mar 12 18:41:16 2019...
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
Error in Vivado_HLS

blinkyang, 2019-03-12:
Also having the same issue with the same error message being thrown.

giuliogamba, 2019-03-13:
Which Vivado version are you using?

blinkyang, 2019-03-14:
I'm using Vivado version 2018.2, but im only using the 30 day trail version, would that make a difference?

BrunoOliveira8, 2019-03-14:
Hello,
Same issue, any suggestion?

Thank you :)

giuliogamba, 2019-03-14:
I'm not able to replicate, hardware generation works under RedHat6 with 2018.2. Cna you please attach the log of your error?

blinkyang, 2019-03-14:
As far as I can tell that's the only log of the error that was given when I tried to execute it, but ill give the whole log of the execution if I missed something. Thanks in advanced for the help though its really appreciated.

xilinx-tiny-cnn already cloned
Calling Vivado HLS for hardware synthesis...

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.3 (64-bit)
  **** SW Build 2405991 on Thu Dec  6 23:36:41 MST 2018
  **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source /home/blinkyang/Documents/AI/Vivado/2018.3/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/home/blinkyang/Documents/AI/Vivado/2018.3/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'blinkyang' on host 'blinkyang-VirtualBox' (Linux_x86_64 version 4.15.0-46-generic) on Thu Mar 14 13:52:58 EDT 2019
INFO: [HLS 200-10] On os Ubuntu 18.04.2 LTS
INFO: [HLS 200-10] In directory '/home/blinkyang/Documents/AI/BNN-PYNQ-master/bnn/src/network/output/hls-syn'
HLS project: cnvW2A2-pynqZ1-Z2
HW source dir: /home/blinkyang/Documents/AI/BNN-PYNQ-master/bnn/src/network/cnvW2A2/hw
BNN HLS library: /home/blinkyang/Documents/AI/BNN-PYNQ-master/bnn/src/library/hls
INFO: [HLS 200-10] Opening project '/home/blinkyang/Documents/AI/BNN-PYNQ-master/bnn/src/network/output/hls-syn/cnvW2A2-pynqZ1-Z2'.
INFO: [HLS 200-10] Adding design file '/home/blinkyang/Documents/AI/BNN-PYNQ-master/bnn/src/network/cnvW2A2/hw/top.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/blinkyang/Documents/AI/BNN-PYNQ-master/bnn/src/network/cnvW2A2/hw/../sw/main_python.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/blinkyang/Documents/AI/BNN-PYNQ-master/bnn/src/library/host/foldedmv-offload.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/blinkyang/Documents/AI/BNN-PYNQ-master/bnn/src/library/host/rawhls-offload.cpp' to the project
INFO: [HLS 200-10] Opening solution '/home/blinkyang/Documents/AI/BNN-PYNQ-master/bnn/src/network/output/hls-syn/cnvW2A2-pynqZ1-Z2/sol1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg400-1'
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch CLANG as the compiler.
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Thu Mar 14 13:53:03 2019...
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
Error in Vivado_HLS

blinkyang, 2019-03-16:
Sorry for all the issues, but I finally got it resolved. After retrying it multiple tiimes, it ended up just being a picky root variable. I ended up setting XILINX_BNN_ROOT to the directory again, but instead of using an indirect reference with "~" i did a direct reference from the home directory and it worked. I'm really grateful for all the help though.

paulr92v, 2019-03-19:
Hi @blinkyang could you please elaborate more on your solution?
I tried your fix(no ~ in the env var) but still having the same error.

>>echo $XILINX_BNN_ROOT
>>/home/work/bnn/bnn/src/

Environment is Description:	Ubuntu 18.04.1 LTS
Shell verison: GNU bash, version 4.4.19(1)-release (x86_64-pc-linux-gnu)

giuliogamba, 2019-03-19:
Hi @paulr92v, can you elaborate on your error? Is this happening right after a fresh clone?

paulr92v, 2019-03-19:
Hi, Yes it is happening after fresh clone.The fact is that the vivado hls synthesis gave me the same error even from the gui. Actually I think I found the problem: Ubuntu 18.04.1. I tried on Ubuntu 16.04.4 and working correctly.

giuliogamba, 2019-03-19:
Hi, 
actually Ubuntu 18.04 is not among the [supported OS](https://www.xilinx.com/products/design-tools/vivado/vivado-webpack.html#os_support) for Vivado Design Suite.

paulr92v, 2019-03-19:
Yes, you are right. Thanks!
Paul

blinkyang, 2019-03-19:
I got it working on Ubuntu 18.04.2 LTS though which is weird, but I just now remembered that I had a fresh Ubuntu OS too, so I also installed build-essentials right before I tried changing the XILINX_BNN_ROOT so that may also be the issue too, but if it works great!

giuliogamba, 2019-03-20:
Ok, seems the problem has been solved. Closing the issue now

closed 2019-03-20 by giuliogamba