Xilinx/BNN-PYNQ #90: Object Localization and Detection with new my own dataset

HungSoma, 2019-03-30:
I trained a my own dataset to detect car and i tested on Ultra96 via PYNQ 2.4 . The result was good but when I tried to follow the sample code on road-traffic , my code could not bound box object (car) good enough. And I want to know how I can modify your code and how to bound box object as well. Finally I want to bound box the car fast enough to be able use for real time .
[image]

This is my bounding box  ( tried to use non-maximum)
Sorry about my English and my unrelated content!
Thanks all!

giuliogamba, 2019-04-02:
Hi, 

the network you trained is for image classification, not for detection. In out notebook examples, we show how it's possible to use an image classification neural network to detect objects, but the accuracy heavily depends on how many tiles (portion of the original image) are fed to the classification neural network. 
In our [example](https://github.com/Xilinx/BNN-PYNQ/blob/master/notebooks/CNV-BNN_Road-Signs.ipynb) we used `for s in [64,96]:` meaning we take 64x64 and 96x96 pixel tiles, with a distance from each other of `stride = s // 4` 1/4 of the image size. 
These values have to be carefully selected, depending on the smaller and the bigger car you want to identify in the original image. 
Please note that the road-sign neural network has been trained with an extra "junk" class, as explained [here](https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/src/training), using background images to be classified as a different class. This approach actually increased the perceived accuracy of the object detection task. 
Hope this helps

HungSoma, 2019-04-05:
Thank you so much , I think I have some ideas to do and improve my junk class. Your advice is exactly what i need. This help me a lot !
Once again, Thanks you!

closed 2019-04-05 by giuliogamba