Xilinx/BNN-PYNQ #92: ERROR: [SIM 211-100] CSim failed with errors

maunns, 2019-04-15:
Hi, This issue was created by someone else before but it wasn't answered. I was hoping, if you could resolve this. Please find below the log of error.

$ ./make-hw.sh cnvW1A2 ultra96 a
xilinx-tiny-cnn already cloned
DEBUGGING #1
Calling Vivado HLS for hardware synthesis...
DEBUGGING #2
vivado_hls -f E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/hls-syn.tcl -tclargs cnvW1A2-ultra96 E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/cnvW1A2/hw E:/masters/special-problem/BNN-PYNQ-master/bnn/src/../params/cifar10/cnvW1A2 E:/masters/special-problem/BNN-PYNQ-master/bnn/src/../../tests/Test_image/deer.bin 4 xczu3eg-sbva484-1-i 3

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.3 (64-bit)
  **** SW Build 2405991 on Thu Dec  6 23:38:27 MST 2018
  **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source E:/Vivado/2018.3/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running 'E:/Vivado/2018.3/bin/unwrapped/win64.o/vivado_hls.exe'
INFO: [HLS 200-10] For user 'Muhammad Shabbir' on host 'm-shabbir' (Windows NT_amd64 version 6.2) on Mon Apr 15 16:40:26 -0400 2019
INFO: [HLS 200-10] In directory 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/output/hls-syn'
HLS project: cnvW1A2-ultra96
HW source dir: E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/cnvW1A2/hw
BNN HLS library: E:/masters/special-problem/BNN-PYNQ-master/bnn/src/library/hls
INFO: [HLS 200-10] Opening project 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/output/hls-syn/cnvW1A2-ultra96'.
INFO: [HLS 200-10] Adding design file 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/cnvW1A2/hw/top.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/cnvW1A2/hw/../sw/main_python.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/library/host/foldedmv-offload.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/library/host/rawhls-offload.cpp' to the project
INFO: [HLS 200-10] Opening solution 'E:/masters/special-problem/BNN-PYNQ-master/bnn/src/network/output/hls-syn/cnvW1A2-ultra96/sol1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 3ns.
INFO: [HLS 200-10] Setting target device to 'xczu3eg-sbva484-1-i'
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch CLANG as the compiler.
make: 'csim.exe' is up to date.
Setting network weights and thresholds in accelerator...
@E Simulation failed.
ERROR: [SIM 211-100] CSim failed with errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Mon Apr 15 16:40:33 2019...
DEBUGGING #3
ERROR: [SIM 211-100] CSim failed with errors.
Error in Vivado_HLS

giuliogamba, 2019-04-16:
On which OS are you having this problem?

aunns, 2019-04-16:
@giuliogamba  I am using windows 10 and I am using git bash to run the shell script.
Please note that my vivado suite is not installed in C drive.
So i had to change the path of vivado and vivado hls in make-hw.sh script to the path that exists on my system.

giuliogamba, 2019-04-17:
Can you please try the hardware build flow on a linux machine?

aunns, 2019-04-21:
@giuliogamba  linux machine solved the problem
Thanks

closed 2019-04-23 by giuliogamba

Changyiyu, 2019-11-28:
@giuliogamba Hi, I have a similar problem. How can I solve it?

$ ./make-hw.sh cnvW1A1 pynqZ1-Z2 a
xilinx-tiny-cnn already cloned
Cloning into 'finn-hlslib'...
remote: Enumerating objects: 136, done.
remote: Counting objects: 100% (136/136), done.
remote: Compressing objects: 100% (73/73), done.
remote: Total 136 (delta 78), reused 112 (delta 60), pack-reused 0
Receiving objects: 100% (136/136), 95.97 KiB | 0 bytes/s, done.
Resolving deltas: 100% (78/78), done.
Checking connectivity... done.
Calling Vivado HLS for hardware synthesis...

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2017.4 (64-bit)
  **** SW Build 2086221 on Fri Dec 15 20:54:30 MST 2017
  **** IP Build 2085800 on Fri Dec 15 22:25:07 MST 2017
    ** Copyright 1986-2017 Xilinx, Inc. All Rights Reserved.

source /opt/Xilinx/Vivado/2017.4/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/opt/Xilinx/Vivado/2017.4/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'takoesl' on host 'takoesl-virtual-machine' (Linux_x86_64 version 4.15.0-70-generic) on Thu Nov 28 22:41:54 CST 2019
INFO: [HLS 200-10] On os Ubuntu 16.04.6 LTS
INFO: [HLS 200-10] In directory '/home/takoesl/Downloads/BNN-PYNQ-master/bnn/src/network/output/hls-syn'
HLS project: cnvW1A1-pynqZ1-Z2
HW source dir: /home/takoesl/Downloads/BNN-PYNQ-master/bnn/src/network/cnvW1A1/hw
BNN HLS library: /home/takoesl/Downloads/BNN-PYNQ-master/bnn/src/library/finn-hlslib
INFO: [HLS 200-10] Opening project '/home/takoesl/Downloads/BNN-PYNQ-master/bnn/src/network/output/hls-syn/cnvW1A1-pynqZ1-Z2'.
WARNING: [HLS 200-40] Cannot find design file '/home/takoesl/Downloads/BNN-PYNQ-master/bnn/src/network/cnvW1A1/hw/top.cpp'
WARNING: [HLS 200-40] Cannot find test bench file '/home/takoesl/Downloads/BNN-PYNQ-master/bnn/src/network/cnvW1A1/hw/../sw/main_python.cpp'
INFO: [HLS 200-10] Adding test bench file '/home/takoesl/Downloads/BNN-PYNQ-master/bnn/src/library/host/foldedmv-offload.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/takoesl/Downloads/BNN-PYNQ-master/bnn/src/library/host/rawhls-offload.cpp' to the project
INFO: [HLS 200-10] Opening solution '/home/takoesl/Downloads/BNN-PYNQ-master/bnn/src/network/output/hls-syn/cnvW1A1-pynqZ1-Z2/sol1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg400-1'
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch CLANG as the compiler.
   Compiling ../../../../../../../library/host/rawhls-offload.cpp in debug mode
   Compiling ../../../../../../../library/host/foldedmv-offload.cpp in debug mode
   Generating csim.exe
Makefile.rules:374: recipe for target 'csim.exe' failed
ERROR: [SIM 211-102] C simulation failed, no 'main' function.
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Thu Nov 28 22:42:00 2019...
ERROR: [SIM 211-102] C simulation failed, no 'main' function.
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
Error in Vivado_HLS

Voltazen, 2021-11-05:
Hello! I have the same problem. It's work probably on linux. But if I work on Windows, how to do to solve it?

mustafatasci42, 2021-11-30:
Hi, I have a same problem. How can I solve it?

mustafa@mustafat:~/Documents/BNN-PYNQ/bnn/src/network$ ./make-hw.sh lfcW1A1 pynqZ1-Z2 a
xilinx-tiny-cnn already cloned
FINN hls library already cloned
Calling Vivado HLS for hardware synthesis...

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.3 (64-bit)
  **** SW Build 2405991 on Thu Dec  6 23:36:41 MST 2018
  **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source /tools/Xilinx/Vivado/2018.3/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/tools/Xilinx/Vivado/2018.3/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'mustafa' on host 'mustafat' (Linux_x86_64 version 5.4.0-90-generic) on Tue Nov 30 21:47:20 +03 2021
INFO: [HLS 200-10] On os Ubuntu 18.04.2 LTS
INFO: [HLS 200-10] In directory '/home/mustafa/Documents/BNN-PYNQ/bnn/src/network/output/hls-syn'
HLS project: lfcW1A1-pynqZ1-Z2
HW source dir: /home/mustafa/Documents/BNN-PYNQ/bnn/src/network/lfcW1A1/hw
BNN HLS library: /home/mustafa/Documents/BNN-PYNQ/bnn/src/library/finn-hlslib
INFO: [HLS 200-10] Creating and opening project '/home/mustafa/Documents/BNN-PYNQ/bnn/src/network/output/hls-syn/lfcW1A1-pynqZ1-Z2'.
INFO: [HLS 200-10] Adding design file '/home/mustafa/Documents/BNN-PYNQ/bnn/src/network/lfcW1A1/hw/top.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/mustafa/Documents/BNN-PYNQ/bnn/src/network/lfcW1A1/hw/../sw/main_python.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/mustafa/Documents/BNN-PYNQ/bnn/src/library/host/foldedmv-offload.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/mustafa/Documents/BNN-PYNQ/bnn/src/library/host/rawhls-offload.cpp' to the project
INFO: [HLS 200-10] Creating and opening solution '/home/mustafa/Documents/BNN-PYNQ/bnn/src/network/output/hls-syn/lfcW1A1-pynqZ1-Z2/sol1'.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg400-1'
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Tue Nov 30 21:47:20 2021...
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
Error in Vivado_HLS