Xilinx/BNN-PYNQ #95: Error When training starts

vesaalexandru95, 2019-05-07:
Hello, I have a problem.
I don't know what to do.Thanks a lot.
Training...
Traceback (most recent call last):
  File "bnn_script.py", line 241, in <module>
    shuffle_parts=shuffle_parts)
  File "C:\Users\G-ro01.dlmtraining\Desktop\BNN_THEANO_TENSORFLOW\BNN-PYNQ\bnn\src\training\quantized_net.py", line 288, in train
    train_loss = train_epoch(X_train,y_train,LR,rotations,oversize_pixels)
  File "C:\Users\G-ro01.dlmtraining\Desktop\BNN_THEANO_TENSORFLOW\BNN-PYNQ\bnn\src\training\quantized_net.py", line 238, in train_epoch
    loss += train_fn(X[i*batch_size:(i+1)*batch_size],y[i*batch_size:(i+1)*batch_size],LR)
  File "c:\users\g-ro01.dlmtraining\desktop\bnn_theano_tensorflow\theano\theano\compile\function_module.py", line 903, in __call__
    self.fn() if output_subset is None else\
  File "c:\users\g-ro01.dlmtraining\desktop\bnn_theano_tensorflow\theano\theano\gof\vm.py", line 305, in __call__
    link.raise_with_op(node, thunk)
  File "c:\users\g-ro01.dlmtraining\desktop\bnn_theano_tensorflow\theano\theano\gof\link.py", line 325, in raise_with_op
    reraise(exc_type, exc_value, exc_trace)
  File "c:\users\g-ro01.dlmtraining\desktop\bnn_theano_tensorflow\theano\theano\gof\vm.py", line 301, in __call__
    thunk()
  File "c:\users\g-ro01.dlmtraining\desktop\bnn_theano_tensorflow\theano\theano\gof\op.py", line 892, in rval
    r = p(n, [x[0] for x in i], o)
  File "c:\users\g-ro01.dlmtraining\desktop\bnn_theano_tensorflow\theano\theano\tensor\elemwise.py", line 790, in perform
    variables = ufunc(*ufunc_args, **ufunc_kwargs)
  File "c:\users\g-ro01.dlmtraining\desktop\bnn_theano_tensorflow\theano\theano\scalar\basic.py", line 1076, in impl
    self.__class__.__name__)
theano.gof.utils.MethodNotDefined: ('impl', <class 'quantization.Round3'>, 'Round3')
Apply node that caused the error: Elemwise{round3}(Elemwise{mul,no_inplace}.0)
Toposort index: 253
Inputs types: [TensorType(float32, 4D)]
Inputs shapes: [(50L, 3L, 32L, 32L)]
Inputs strides: [(12288L, 4096L, 128L, 4L)]
Inputs values: ['not shown']
Outputs clients: [[Elemwise{true_div,no_inplace}(Elemwise{round3}.0, InplaceDimShuffle{x,x,x,x}.0)]]

Backtrace when the node is created(use Theano flag traceback.limit=N to make it longer):
  File "bnn_script.py", line 203, in <module>
    train_output = lasagne.layers.get_output(cnn, deterministic=False)
  File "c:\users\g-ro01.dlmtraining\desktop\bnn_theano_tensorflow\lasagne\lasagne\layers\helper.py", line 197, in get_output
    all_outputs[layer] = layer.get_output_for(layer_inputs, **kwargs)
  File "c:\users\g-ro01.dlmtraining\desktop\bnn_theano_tensorflow\lasagne\lasagne\layers\special.py", line 52, in get_output_for
    return self.nonlinearity(input)
  File "C:\Users\G-ro01.dlmtraining\Desktop\BNN_THEANO_TENSORFLOW\BNN-PYNQ\bnn\src\training\quantized_net.py", line 59, in __call__
    return self.quantization.quantize(y)
  File "C:\Users\G-ro01.dlmtraining\Desktop\BNN_THEANO_TENSORFLOW\BNN-PYNQ\bnn\src\training\quantization.py", line 145, in quantize
    return T.cast(round3((T.clip(X, self.min, self.max) + self.shift)*self.scale) / self.scale - self.shift, theano.config.floatX)

HINT: Use the Theano flag 'exception_verbosity=high' for a debugprint and storage map footprint of this apply node.

nickfraser, 2019-05-07:
This is strange, can you answer me the following questions:
1. It looks like you're using your own training script ('bnn_script.py'), do you get the same error when you run one of our provided examples? For example 'mnist.py' or 'cifar10.py'?
2. What versions of Theano & Lasagne are you using?

vesaalexandru95, 2019-05-07:
Yes. I use my own dataset. But on Linux for example it works. I.m using
theano 0.8.2 and lasagne 0.2dev1 but i tried also with another versions and
same error on windows. Is very strange

nickfraser, 2019-05-08:
I'm sorry, windows is an unsupported platform for the training scripts. If you want to run this on windows, please use docker and our provided [Dockerfiles](https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/src/training/docker). If you decide to continue to try to run this natively on windows, good luck!

Also, the supported version of Theano is 0.9.0beta1.

closed 2019-05-08 by nickfraser

vesaalexandru95, 2019-05-08:
Ok. I tried on another machine with linux and I get this error.
Please help.
Thanks
(base) autoliv@safetydance:~/Desktop/BNN-PYNQ/bnn/src/training$ python mnist.py
activation_bits = 1
weight_bits = 1
batch_size = 100
alpha = 0.1
epsilon = 0.0001
num_epochs = 1000
dropout_in = 0.2
dropout_hidden = 0.5
W_LR_scale = Glorot
LR_start = 0.003
LR_fin = 3e-07
LR_decay = 0.990831944893
save_path = mnist-1w-1a.npz
shuffle_parts = 1
Loading MNIST dataset...
Building the MLP...
/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/tensor/basic.py:2146: UserWarning: theano.tensor.round() changed its default from `half_away_from_zero` to `half_to_even` to have the same default as NumPy. Use the Theano flag `warn.round=False` to disable this warning.
  "theano.tensor.round() changed its default from"
W_LR_scale = 34.7179
H = 1.0
W_LR_scale = 36.9504
H = 1.0
W_LR_scale = 36.9504
H = 1.0
W_LR_scale = 26.2552
H = 1.0
WARNING (theano.tensor.blas): We did not found a dynamic library into the library_dir of the library we use for blas. If you use ATLAS, make sure to compile it with dynamics library.
===============================
00001	#include <Python.h>
00002	#include <iostream>
00003	#include "theano_mod_helper.h"
00004	#include <math.h>
00005	#include <numpy/arrayobject.h>
00006	#include <numpy/arrayscalars.h>
00007	#include <vector>
00008	#include <algorithm>
00009	//////////////////////
00010	////  Support Code
00011	//////////////////////
00012	
00013	    namespace {
00014	    struct __struct_compiled_op_fb25a0bdee2d90f52cade6e6044e552d {
00015	        PyObject* __ERROR;
00016	
00017	        PyObject* storage_V3;
00018	PyObject* storage_V5;
00019	PyObject* storage_V7;
00020	PyObject* storage_V9;
00021	PyObject* storage_V11;
00022	PyObject* storage_V13;
00023	PyObject* storage_V1;
00024	        
00025	
00026	        __struct_compiled_op_fb25a0bdee2d90f52cade6e6044e552d() {
00027	            // This is only somewhat safe because we:
00028	            //  1) Are not a virtual class
00029	            //  2) Do not use any virtual classes in the members
00030	            //  3) Deal with mostly POD and pointers
00031	
00032	            // If this changes, we would have to revise this, but for
00033	            // now I am tired of chasing segfaults because
00034	            // initialization code had an error and some pointer has
00035	            // a junk value.
00036	            memset(this, 0, sizeof(*this));
00037	        }
00038	        ~__struct_compiled_op_fb25a0bdee2d90f52cade6e6044e552d(void) {
[... 1317 lines cut ...]
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/compile/function_module.py", line 1795, in orig_function
    defaults)
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/compile/function_module.py", line 1661, in create
    input_storage=input_storage_lists, storage_map=storage_map)
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/gof/link.py", line 699, in make_thunk
    storage_map=storage_map)[:3]
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/gof/vm.py", line 1047, in make_all
    impl=impl))
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/gof/op.py", line 935, in make_thunk
    no_recycling)
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/gof/op.py", line 839, in make_c_thunk
    output_storage=node_output_storage)
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/gof/cc.py", line 1190, in make_thunk
    keep_lock=keep_lock)
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/gof/cc.py", line 1131, in __compile__
    keep_lock=keep_lock)
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/gof/cc.py", line 1586, in cthunk_factory
    key=key, lnk=self, keep_lock=keep_lock)
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/gof/cmodule.py", line 1159, in module_from_key
    module = lnk.compile_cmodule(location)
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/gof/cc.py", line 1489, in compile_cmodule
    preargs=preargs)
  File "/home/autoliv/anaconda2/lib/python2.7/site-packages/theano/gof/cmodule.py", line 2316, in compile_str
    (status, compile_stderr.replace('\n', '. ')))
Exception: ('The following error happened while compiling the node', Elemwise{Composite{(i0 * (i1 + (i0 * round3(clip(i2, i3, i4)))) * i5)}}[(0, 2)](TensorConstant{(1, 1) of 2.0}, TensorConstant{(1, 1) of -1.0}, Elemwise{Composite{(i0 * (i1 + clip(i2, i3, i4)))}}[(0, 2)].0, TensorConstant{(1, 1) of 0}, TensorConstant{(1, 1) of 1}, Elemwise{Composite{Cast{float64}(LT(i0, i1))}}[(0, 0)].0), '\n', "Compilation failed (return status=1): /home/autoliv/.theano/compiledir_Linux-4.10--generic-x86_64-with-debian-stretch-sid-x86_64-2.7.16-64/tmpSeCPeV/mod.cpp:954:2: warning: character constant too long for its type [enabled by default].  ['V15_tmp2'] = round(['V15_tmp1']);.   ^. /home/autoliv/.theano/compiledir_Linux-4.10--generic-x86_64-with-debian-stretch-sid-x86_64-2.7.16-64/tmpSeCPeV/mod.cpp:954:23: warning: character constant too long for its type [enabled by default].  ['V15_tmp2'] = round(['V15_tmp1']);.                        ^. /home/autoliv/.theano/compiledir_Linux-4.10--generic-x86_64-with-debian-stretch-sid-x86_64-2.7.16-64/tmpSeCPeV/mod.cpp:1076:2: warning: character constant too long for its type [enabled by default].  ['V15_tmp2'] = round(['V15_tmp1']);.   ^. /home/autoliv/.theano/compiledir_Linux-4.10--generic-x86 [... cut ...] xe2\x80\x98{anonymous}::__struct_compiled_op_fb25a0bdee2d90f52cade6e6044e552d::run()::__lambda3\xe2\x80\x99 to \xe2\x80\x98double\xe2\x80\x99 for argument \xe2\x80\x981\xe2\x80\x99 to \xe2\x80\x98double round(double)\xe2\x80\x99. ", '[Elemwise{Composite{(i0 * (i1 + (i0 * round3(clip(i2, i3, i4)))) * i5)}}[(0, 2)](TensorConstant{(1, 1) of 2.0}, TensorConstant{(1, 1) of -1.0}, <TensorType(float64, matrix)>, TensorConstant{(1, 1) of 0}, TensorConstant{(1, 1) of 1}, <TensorType(float64, matrix)>)]')