Xilinx/BNN-PYNQ #96: Error when training the BNN

vesaalexandru95, 2019-05-09:
Hello, I come with a new problem.I ran the code on a Virtual Machine with Ubuntu 17.04.
I allocated 13 gb to this machine. I've installed all the dependecies provided in the readme file.
Here is the gist related to the problem : https://gist.github.com/vesaalexandru95/840fb21cc4d851901ca995b54df4186f#file-gistfile1-txt

Thanks a lot,

nickfraser, 2019-05-09:
It looks like there is an issue with the way that Theano is generating the C++ code when [this code snippet](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/training/quantization.py#L46) is injected. Can you post the output of the following from the python console?

>>> import theano
>>> theano.__version__

For reference, I get the string '0.9.0beta1.dev-51ac3abd047e5ee4630fa6749f499064266ee164'

In the meantime, have you tried building the docker image? If so, do you get the same error when using the docker image?

vesaalexandru95, 2019-05-10:
OK . I solved the problem  also in docker and virtual machine. The problem was on some script modified by me.
Thanks a lot for the support.

closed 2019-05-10 by vesaalexandru95