Xilinx/BNN-PYNQ #99: BNN image pre-processing on hardware

Nomi070, 2019-05-16:
I have made a image pre-processor ip core which is directly writing the cifar10 based data into the DDR. And from the DDR i have successfully read all the data in linux running on ARM and write to a text file. Now i want to bypass crop resize and flatten and provide that data directly to the BNN hardware accelerator. I am not very good in python. So i am looking for some suggestions to how can i achieve my task and how the hardware accelerator is receiving the cifar10 data.

maltanar, 2019-05-22:
Hi Nomi070, is your question about how to integrate your preprocessing hardware with the BNN, or (after integrating the hardware) how to modify the software so that it doesn't go through the software preprocessing but provides the hardware with raw images instead?

Nomi070, 2019-05-22:
I want to modify the software. My hardware ip is providing me cifar10 format data. I have done some digging correct me if i am wrong. Cropped images are appended together in a tile format and then pass to classify_images.  And in classify_image image_to_cifar is called. and in image_to_cifar data is written to some file fp. Should i write my data directly to this file. If yes how much data.? 3073 bytes.? and which signal or variable should i wait for to check if there is a stop sign detected or not.?

Nomi070, 2019-05-28:
@maltanar Hello, Need your help, waiting for your reply.

maltanar, 2019-05-28:
We're actually in the process of making a better-documented release in the coming months, but until that time this will take some manual digging through the code to achieve. The Python interface (here for the CNV-w1a1 overlay) calls the C++ host code, which in turns calls the driver to talk to the hardware. If you are more comfortable with C/C++ I suggest taking a look at those files:

https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/cnvW1A1/sw/main_python.cpp
...which in turn calls:
https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/library/host/foldedmv-offload.cpp
https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/library/host/foldedmv-offload.h
...which calls the driver to perform register accesses and copy buffers back and forth:
https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/library/driver/xlnkdriver.hpp

You can see how we check if the accelerator is finished in the ExecAccel function here:
https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/library/host/foldedmv-offload.cpp#L381

You may also find the diagrams in this page helpful (Google Translate may help if you don't speak Japanese):
https://tkat0.github.io/study-BNN-PYNQ/soft-overview.html#

By the way, if you are familiar with hardware design using Vivado HLS, you may want to change your hardware so that its output is streamed directly into the BNN accelerator instead of going first into DRAM and then back out. This should give lower latency and higher performance. If you want to try this, have a look at e.g. the DoCompute function here to see how we construct the dataflow pipeline:

https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/cnvW1A1/hw/top.cpp#L137

Nomi070, 2019-12-20:
@maltanar Did you guys release any more document with the detailed explanation of the the both hardware and software part?

maltanar, 2019-12-20:
We're working on a complete redesign of FINN at the moment, it's actually in public development here:

https://github.com/Xilinx/finn/tree/dev/

There's now a project website as well at 

http://xilinx.github.io/finn

We don't have a release for this new version yet although it's getting there, hopefully in January. The demos in BNN-PYNQ will eventually be built using this new version of the FINN compiler.

closed 2020-09-30 by giuliogamba