Xilinx/BNN-PYNQ #101: Is it a bug related to padding?

HyunwooKim2, 2019-06-03:
Hello
There seems to be a bug related to padding in foldedmv-offload.h of the host code.
I linked the source code below.

https://github.com/Xilinx/BNN-PYNQ/blob/32eed91994228d1042a16c692047dfe81e47d498/bnn/src/library/host/foldedmv-offload.h#L251

When initializing the interleaver, it is initialized to 32 * 32.
I think the pad size should be considered when padding.
Please let me know if I have overlooked it.
Thank you.

giuliogamba, 2019-06-06:
The input images are not padded, they're 32x32 pixels, with 3 channels RGB and 8 bit per channel. The interleaver has the number of channels as first template parameter (3) and the size as the second (32x32).

HyunwooKim2, 2019-06-06:
So you mean padding is not supported?
I am modifying the source somewhat on my local for input padding.
Thank you for your answer.

closed 2019-12-06 by giuliogamba