Xilinx/BNN-PYNQ #103: TMEM in config.h

raminrasoulinezhad, 2019-06-10:
Hi, 

I guess something is wrong with TMEM for convL2 and convL3 in config.h file ( BNN-PYNQ/bnn/src/network/cnvW1A1/hw/config.h ). Could you please check it? Thanks.

Am I right? "OFM_CH = PE * TMEM"?

giuliogamba, 2019-06-17:
No, the values as pushed in the repo are correct. 
`OFM_CH*KERNEL*KERNEL = PE*TMEM`.

closed 2019-06-17 by giuliogamba

raminrasoulinezhad, 2019-07-02:
Hi @giuliogamba 

So, There no streaming between the layers. Am I right? 
As I understood, each layer produces the whole output feature map by getting the whole input feature map. Then, using double buffering, the output is passed to the next layer. Am I right?

Thanks a lot, 
Sincerely

giuliogamba, 2019-07-02:
No, that's not correct. Each layer indeed computes the whole feature map by using the input feature map, but layers communicates via streams enabling concurrent execution of multiple layers (since one layer starts computing as soon as the previous layer output enough data for it to start computing). 
You might want to read the original paper [here ](https://arxiv.org/pdf/1612.07119.pdf), specifically Figure 2.

raminrasoulinezhad, 2019-07-02:
Thanks, 

And just one more question: 
SIMD size is 32. Could you please help me and tell me what was the bottleneck which prevents you to set it to 64 or larger? CLB usage? Memory blocks? Memory interface?

Thanks

giuliogamba, 2019-07-02:
Sometimes SIMD is 32, sometimes is smaller or higher. In the current release the maximum value is 64, and it's a limit added because the parameters are updated via AXI-Lite (which is 64-bit wide in this project). This limit can be removed by modifying the way in which the parameter are updated in the accelerator.