Xilinx/BNN-PYNQ #105: Error during CoSimulation

PascalBACCHUS, 2019-07-05:
Hi,

I'm using Vivado 2019.1 and the lfcW1A1 configuration provided in the network folder.
All the errors are due to templates/variables/scope not found in foldedmv-offload.h, but I already checked the path of all the source/testbench as in hls-syn.tcl and every file are included.

I attach the error file:
[Error.txt](https://github.com/Xilinx/BNN-PYNQ/files/3363231/Error.txt)
Thanks,

giuliogamba, 2019-07-09:
We are able to reproduce the problem, and we are working to fix it. 
In the meantime, I would suggest to use the suggested Vivado version (2018.2)

PascalBACCHUS, 2019-07-15:
Hi,

It works on my laptop which has version 2018.3 but I get this error message on my desktop even though it's also working on 2018.3:

ERROR: System recieved a signal named SIGSEGV and the program has to stop immediately!
This signal was generated when a program tries to read or write outside the memory that is allocated for it, or to write memory that can only be read.
Possible cause of this problem may be: 1) the depth setting of pointer type argument is much larger than it needed; 2)insufficient depth of array argument; 3)null pointer etc.
Current execution stopped during CodeState = ENTER_WRAPC_PC.
You can search CodeState variable name in apatb*.cpp file under ./sim/wrapc dir to locate the position.

ERROR: [COSIM 212-379] Detected segmentation violation, please check C tb.
ERROR: [COSIM 212-362] Aborting co-simulation: C TB post check failed.

closed 2020-09-30 by giuliogamba