Xilinx/BNN-PYNQ #107: I have meet some problems when synthsis the code in Vivado HLS 2018.3

hywhjj, 2019-07-09:
I have meet some problems when synthsis the code in Vivado HLS 2018.3,the cflags is -std=gnu++0x,and meet some errors.

ERROR: [HLS 200-70] Compilation errors found: In file included from BNN_test/src/top.cpp:1:
In file included from BNN_test/src/top.cpp:44:
In file included from BNN_test/src/bnn-library.h:49:
In file included from D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\iostream:39:
In file included from D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ostream:39:
In file included from D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ios:39:
In file included from D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception:151:
D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception_ptr.h:132:13: error: unknown type name 'type_info'
      const type_info*
            ^
In file included from BNN_test/src/top.cpp:1:
In file included from BNN_test/src/top.cpp:44:
In file included from BNN_test/src/bnn-library.h:49:
In file included from D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\iostream:39:
In file included from D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ostream:39:
In file included from D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ios:39:
In file included from D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception:151:
D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:62:5: error: the parameter for this explicitly-defaulted copy constructor is const, but a member or base requires it to be non-const
    nested_exception(const nested_exception&) = default;
    ^
D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:64:23: error: the parameter for this explicitly-defaulted copy assignment operator is const, but a member or base requires it to be non-const
    nested_exception& operator=(const nested_exception&) = default;
                      ^
D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:77:28: error: exception specification in declaration does not match previous declaration
  inline nested_exception::~nested_exception() = default;
                           ^
D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:66:20: note: previous declaration is here
    inline virtual ~nested_exception();
                   ^
D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:122:61: error: redefinition of default argument
    __throw_with_nested(_Ex&& __ex, const nested_exception* = 0)
                                                            ^ ~
D:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:110:56: note: previous definition is here
    __throw_with_nested(_Ex&&, const nested_exception* = 0)
                                                       ^ ~
5 errors generated.
Failed during preprocessing.
    while executing
"source G:/work/work_verilog/work_Vivado/HLS/BNN_PYNQ/BNN_test/solution1/csynth.tcl"
    invoked from within
"hls::main G:/work/work_verilog/work_Vivado/HLS/BNN_PYNQ/BNN_test/solution1/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished C synthesis.
How can I do to slove this problem?

giuliogamba, 2019-07-11:
It seems a problem in your windows installation. Do you have the same problem with the flag `-std=c++0x`?

hywhjj, 2019-07-11:
YES,l meet the same problem when setting the flag -std=c++0x

hywhjj, 2019-07-11:
Now i try to regenerate the project in ubuntu and i set the vivado path in /opt/Xilinx/Vivado/2017.4/bin/unwrapped/lnx64.o
when I put ./make-hw.sh cnvW1A1 pynqZ1-Z2 a in terminal ,it prompts：
Error: The file  is corrupt. Please re-install this software from the original media.

but I can launch vivado when putiing those in terminal:
source /opt/Xilinx/Vivado/2017.4/settings64.sh
vivado &

so I think it was the wrong path leading to this problem,but when i don't set the path ,it prompts:
 Error: Vivado not found.

So how can i set the correct path?

giuliogamba, 2019-07-11:
You have to execute, in the same terminal:
1. `source /opt/Xilinx/Vivado/2017.4/settings64.sh`
2. `./make-hw.sh cnvW1A1 pynqZ1-Z2 a`

Does this work?

giuliogamba, 2019-11-21:
@hywhjj Any updates about this?

closed 2019-12-06 by giuliogamba