Xilinx/BNN-PYNQ #108: Classification rate CoSim

PascalBACCHUS, 2019-07-21:
Hi,

I'm running Cosimulations for the binarized LFC network to measure the classification rate achieved.
I obtain 272 class/s and even when I increase parallel resources (PE, SIMD), it doesn't go higher than 300.

Is this an expected behaviour? I thought we could attain thousands of classifications per seconds.

PascalBACCHUS, 2019-07-24:
I put a link to the output folder with the Vivado project and reports.
https://drive.google.com/drive/folders/1xtBxns_jo0UX-DBBDQJSLstw4i3bPLYg?usp=sharing
The config.h/top.cpp and weights used were the one from the repository.

giuliogamba, 2019-08-01:
How are you measuring 300 frames/s? 
Cosim executes the classification of a single image, but being a dataflow implementation, the more images you want to classify the higher the throughput. Do you see 300 frames/s also in the generated hardware when deployed?

closed 2020-09-30 by giuliogamba