Xilinx/BNN-PYNQ #109: CNV model use other pixel size of cifar10 in BNN

Changyiyu, 2019-08-01:
Hi,

I want to use the resolution of 240*480 of the picture. Is this feasible?
Thank you for your help.

giuliogamba, 2019-09-17:
Currently the HLS library does not support non-squared images.

Changyiyu, 2019-10-16:
Hi,
I want to use a pixel size larger than 32x32 (such as 480x480), and have modified cnv.py and two cifar10.py, they are located in the training directory and pylearn2, but the error is as follows:

virtual-machine:~/BNN-PYNQ-master/bnn/src/training$ python cifar10.py -wb 1 -ab 1
WARNING (theano.sandbox.cuda): The cuda backend is deprecated and will be removed in the next release.  Please switch to the gpuarray backend. You can get more information about how to switch at this URL:
 https://github.com/Theano/Theano/wiki/Converting-to-the-new-gpu-back-end%28gpuarray%29

ERROR (theano.sandbox.cuda): nvcc compiler not found on $PATH. Check your nvcc installation and try again.
activation_bits = 1
weight_bits = 1
batch_size = 50
alpha = 0.1
epsilon = 0.0001
W_LR_scale = Glorot
num_epochs = 5
LR_start = 0.001
LR_fin = 3e-07
LR_decay = 0.197435048583
save_path = cifar10-1w-1a.npz
train_set_size = 450
shuffle_parts = 1
Loading CIFAR-10 dataset...
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_1
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_2
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_3
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_4
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_5
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/test_batch
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_1
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_2
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_3
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_4
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_5
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/test_batch
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_1
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_2
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_3
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_4
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/data_batch_5
loading file /home/takoesl/.pylearn2/cifar10/cifar-10-batches-py/test_batch
Building the CNN...
Traceback (most recent call last):
  File "cifar10.py", line 161, in <module>
    cnn = cnv.genCnv(input, 5, learning_parameters)
  File "/home/takoesl/BNN-PYNQ-master/bnn/src/training/cnv.py", line 198, in genCnv
    num_units=512)      
  File "/home/takoesl/BNN-PYNQ-master/bnn/src/training/quantized_net.py", line 76, in __init__
    super(DenseLayer, self).__init__(incoming, num_units, W=lasagne.init.Uniform((self.quantization.min,self.quantization.max)), **kwargs)
  File "/home/takoesl/.local/lib/python2.7/site-packages/lasagne/layers/dense.py", line 103, in __init__
    self.W = self.add_param(W, (num_inputs, num_units), name="W")
  File "/home/takoesl/.local/lib/python2.7/site-packages/lasagne/layers/base.py", line 234, in add_param
    param = utils.create_param(spec, shape, name)
  File "/home/takoesl/.local/lib/python2.7/site-packages/lasagne/utils.py", line 376, in create_param
    spec = spec(shape)
  File "/home/takoesl/.local/lib/python2.7/site-packages/lasagne/init.py", line 31, in __call__
    return self.sample(shape)
  File "/home/takoesl/.local/lib/python2.7/site-packages/lasagne/init.py", line 100, in sample
    low=self.range[0], high=self.range[1], size=shape))
  File "mtrand.pyx", line 1568, in mtrand.RandomState.uniform (numpy/random/mtrand/mtrand.c:16695)
  File "mtrand.pyx", line 234, in mtrand.cont2_array_sc (numpy/random/mtrand/mtrand.c:3000)
MemoryError

Do I miss some steps when the pixels get bigger?

nickfraser, 2019-10-17:
Are you able to run the original training scripts (without your modifications)?

Changyiyu, 2019-10-18:
Yes, I can. But I can't train when I change the pixel size.

nickfraser, 2019-10-18:
I see. In that case I believe you are running out of memory (as in RAM), please try again on a system with more memory.

closed 2019-10-18 by nickfraser